margelo/react-native-vision-camera #994: ❓ Full expo support

obasille, 2022-04-14:
### What feature or enhancement are you suggesting?

First, thank you for this great package!

Would it be possible to have the expo config plugin to change `compileSdkVersion` to 31?

I had to manually make that change which breaks the idea of the plugin doing all the required config.

For reference, here are the commands I've run:
```
expo init PixelsValidationApp
expo install react-native-vision-camera@2.13.0
expo run:android
```

Note: I'm using version 2.13.0 because of this issue: https://github.com/mrousavy/react-native-vision-camera/issues/957

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

Manually edit `build.gradle` and set `compileSdkVersion = 31`

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

viljark, 2022-04-20:
I think it is not planned since [different libraries may want different configuration](https://github.com/mrousavy/react-native-vision-camera/pull/991#issuecomment-1096462073) and user should set it themselves with custom plugin: https://github.com/mrousavy/react-native-vision-camera/pull/717#issuecomment-1088091392

mrousavy, 2022-04-20:
Exactly what @viljark said.

closed 2022-04-20 by mrousavy