margelo/react-native-vision-camera #997: 🐛 If the component is updated too many times using the frame processor, the application will crash.

HyopeR, 2022-04-15:
### What were you trying to do?

Hello, I was using frame processors for text recognition with camera and barcode reading with camera. I update my components based on the results of the frame processor. I use the runOnJS and useState methods for this. There is no problem in updating the components. But frame processors throw an exception when I continue this process too much.

This only happens when I try to update components with frame processor results. If I don't update the components, no exception is thrown.

This issue causes the app to crash. adb logcat *:E I detected the error.

### Reproduceable Code

```tsx
import React, {useState} from 'react';
import {StyleSheet, View} from 'react-native';
import {
  Camera,
  useCameraDevices,
  useFrameProcessor,
} from 'react-native-vision-camera';
import {scanOCR} from 'vision-camera-ocr';
import {runOnJS} from 'react-native-reanimated';
import {useIsForeground} from './src/utils/hooks';
import {useIsFocused} from '@react-navigation/native';

const CodeCamera = React.forwardRef(({...props}, ref) => {
  const devices = useCameraDevices('wide-angle-camera');
  const isActive = useIsForeground();
  const isFocused = useIsFocused();

  const device = devices.back;
  const [code, setCode] = useState('');

  const textRecognize = data => {
    const isNotEmpty = data && data?.result?.text !== '';
    isNotEmpty && setCode(data.result.text);
  };

  const frameProcessor = useFrameProcessor(
    frame => {
      'worklet';
      let text = scanOCR(frame);
      runOnJS(textRecognize)(text);
    },
    [textRecognize],
  );
  return (
    <View style={styles.wrapper}>
      {device && isFocused && (
        <Camera
          ref={ref}
          style={styles.camera}
          device={device}
          isActive={isActive}
          frameProcessor={frameProcessor}
          frameProcessorFps={5}
          preset={'vga-640x480'}
          {...props}
        />
      )}
    </View>
  );
});

const styles = StyleSheet.create({
  wrapper: {
    flex: 1,
    position: 'relative',
  },
  camera: {
    flex: 1,
    ...StyleSheet.absoluteFill,
[... 104 lines cut ...]
04-15 19:22:04.248  2559  2559 E audit   : type=1400 audit(1650039724.243:1022): avc:  denied  { search } for  pid=17734 comm="crash_dump64" name="com.google.android.gms" dev="dm-0" ino=204 scontext=u:r:crash_dump:s0:c512,c768 tcontext=u:object_r:app_data_file:s0:c512,c768 tclass=dir permissive=0 SEPF_SM-A520F_8.0.0_0010 audit_filtered
04-15 19:22:04.248  2559  2559 E audit   : type=1300 audit(1650039724.243:1022): arch=c00000b7 syscall=56 success=no exit=-13 a0=ffffff9c a1=734aa30060 a2=0 a3=0 items=0 ppid=1 pid=17734 auid=4294967295 uid=10332 gid=10332 euid=10332 suid=10332 fsuid=10332 egid=10332 sgid=10332 fsgid=10332 tty=(none) ses=4294967295 comm="crash_dump64" exe="/system/bin/crash_dump64" subj=u:r:crash_dump:s0:c512,c768 key=(null)
04-15 19:22:04.248  2559  2559 E audit   : type=1327 audit(1650039724.243:1022): proctitle=63726173685F64756D703634003136393534003137373332
04-15 19:22:04.288  2580 16952 E ExynosCamera3: [CAM_ID(0)][]-[m_monitorThreadFunc] (8069) (0)
04-15 19:22:04.288  2580 16952 E ExynosCamera3: [CAM_ID(0)][]-[m_monitorThreadFunc] (8071) ERROR_DQ_BLOCKED) ; ERROR_DQ_BLOCKED_COUNT =20
04-15 19:22:04.488  2605  2605 E /system/bin/tombstoned: Tombstone written to: /data/tombstones//tombstone_06
04-15 19:22:04.497  2580 16933 E Camera3-OutputStream: getBufferLockedCommon: Stream 1: Can't dequeue next output buffer: Connection timed out (-110)
04-15 19:22:04.497  2580 16933 E Camera3-Device: RequestThread: Can't get output buffer, skipping request: Connection timed out (-110)
04-15 19:22:04.498  2559  2559 E audit   : type=1701 audit(1650039724.493:1023): auid=4294967295 uid=10332 gid=10332 ses=4294967295 subj=u:r:untrusted_app:s0:c512,c768 pid=16954 comm="pool-18-thread-" exe="/system/bin/app_process64" sig=11
```

### Device

Samsung A5 Android 8

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

HyopeR, 2022-04-15:
Also, when a component is updated with frame processors, the following error is constantly printed in the console. This error is not printed when the component is not updated.

PS: I think they are related to each other. That's why I stated that too.

`
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions: Unhandled SoftException
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions: com.facebook.react.bridge.AssertionException: Expected to run on UI thread!
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.facebook.react.bridge.SoftAssertions.assertCondition(SoftAssertions.java:37)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.facebook.react.bridge.UiThreadUtil.assertOnUiThread(UiThreadUtil.java:26)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.facebook.react.uimanager.UIManagerModule.resolveView(UIManagerModule.java:960)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.mrousavy.camera.frameprocessor.FrameProcessorRuntimeManager.findCameraViewById(FrameProcessorRuntimeManager.kt:65)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at android.os.Handler.handleCallback(Handler.java:789)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at android.os.Handler.dispatchMessage(Handler.java:98)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at android.os.Looper.loop(Looper.java:164)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:226)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at java.lang.Thread.run(Thread.java:764)
`

HyopeR, 2022-04-16:
After further research I had a look at this [https://github.com/software-mansion/react-native-reanimated/issues/3011](https://github.com/software-mansion/react-native-reanimated/issues/3011) problem. I discovered that the problem might be on the react-native-reanimated side. I updated my project's dependencies.

**Versions**
<pre>
"react": "17.0.2",
"react-native": "^0.68.1",
"react-native-reanimated": "2.7.0",
"react-native-vision-camera": "^2.13.2",
</pre>

But it didn't help me to get rid of this recurring error in log with all these updates. My thinking is that an exception in the react-native-reanimated package is filling up the data log. And then it causes the app to crash. `com.facebook.react.uimanager.UIManagerModule.resolveView(UIManagerModule.java:960)`

<pre>
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions: Unhandled SoftException
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions: com.facebook.react.bridge.AssertionException: Expected to run on UI thread!
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.facebook.react.bridge.SoftAssertions.assertCondition(SoftAssertions.java:37)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.facebook.react.bridge.UiThreadUtil.assertOnUiThread(UiThreadUtil.java:26)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.facebook.react.uimanager.UIManagerModule.resolveView(UIManagerModule.java:960)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.mrousavy.camera.frameprocessor.FrameProcessorRuntimeManager.findCameraViewById(FrameProcessorRuntimeManager.kt:65)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at android.os.Handler.handleCallback(Handler.java:789)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at android.os.Handler.dispatchMessage(Handler.java:98)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at android.os.Looper.loop(Looper.java:164)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:226)
04-15 19:32:55.497 18815 19469 E unknown:SoftAssertions:        at java.lang.Thread.run(Thread.java:764)
</pre>

HyopeR, 2022-04-16:
If anyone has the same problem as me, **they can do this as a workaround** . I don't get any errors after these adjustments. And the application works successfully without crashing.

**1-** If hermes is active, turn it off. android/app/build.gradle hermes disable.
**2-** Downgraded the vision-camera version. "react-native-vision-camera": "2.13.0".

HyopeR, 2022-04-18:
[https://github.com/mrousavy/react-native-vision-camera/issues/350](https://github.com/mrousavy/react-native-vision-camera/issues/350) It could be an extension of the topic discussed here. However, the current version of the react-native-reanimated package does not prevent crashes.

mrousavy, 2022-04-19:
Looks like #928 is related. I'll try to find a fix when migrating VisionCamera to Fabric.

HyopeR, 2022-04-19:
I don't think the crashing issue is caused by version 2.13.1. Because I tried many times using **react-native-vision-camera version 2.13.0** and **react-native-reanimated 2.3.0-beta.2, 2.3.0, 2.5.0 and 2.7.0.** However, in all my attempts, when I tried to update the component using the frame processor, the application crashed. If Hermes is turned off, there is no problem. Do you think it will be possible to update components using the frame processor with Hermes turned on in future versions?

tcank, 2022-05-09:
Hi everyone. We have the same issue, with hermes enabled and using:

```
"react-native-vision-camera": "2.13.1",
"react-native-reanimated": "2.7.0",
```

Is not an option to us to disable hermes. We only have registries of the issue occurring on Android platforms. I attach some callstacks of users that ends with the SIGSEGV segfault. You can watch that the callstack in this cases are always different. The most of they have a call to `vision::VisionCameraScheduler::triggerUI` but also we found a case without it and it is the one with a call to `vision::JSIJNIConversion::convertJNIObjectToJSIValue`. Maybe, this one is a different issue.

If we can provide more info about our app setup or the error itself, let me know. We are available for any specific debug or test to collect more info and get some fix.

Thanks!

| Case 1 | Case 2 | Case 3 | Case 4 |
|--------|--------|--------|--------|
| [image]     | [image]     | [image]     | [image]     |

quangphuctq2000, 2023-06-15:
I think the reason is camera is capturing frame by real time, too many useFrameProcessor callback => crash. Perhaps open camera only we need to reduce crash times 🤔

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy