margelo/react-native-vision-camera #1004: 🐛 App freezes when toggling to selfie camera

pierroo, 2022-04-18:
### What were you trying to do?

Using the example from this repo, in the CameraPage there is a button to toggle front/back camera.
this button is only displayed if there is indeed both front and back camera (const "supportsCameraFlipping")

Unfortunately, the app freezes when tapping on it on some android devices.
( could not reproduce on mine, but many users complained about it; and I see a LOT of new ANR in my google play console )

### Reproduceable Code

```tsx
use the example code provided by this library.
Tap on the "toggle" camera back/front; experience the freeze. (on some android devices)
```

### What happened instead?

the app does not actually crash, but freeze until we toggle back again to initial default camera.

### Device

Android

### VisionCamera Version

2.13.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-04-19:
Hey, thank your for the issue report!

Unfortunately I cannot reproduce the error. And you did not provide a reproduceable sample code (you're not shipping this example/ app in the app-store are you?). Also, you didn't provide any logs.

Sorry, I know I'm a really good developer but I'm not a magician, don't have any way to reproduce the error.

closed 2022-04-19 by mrousavy

pierroo, 2022-04-19:
Thank you @mrousavy , I fully understand.

Here is the log from the google play crash console:

```
java.lang.IllegalStateException: 
  at c.d.a.a1.a
  at c.d.a.f1.O
  at c.d.a.f1.V
  at c.d.a.f1.T
  at c.d.a.f1.U
  at c.d.a.b0.a
  at androidx.camera.camera2.e.w1.R
  at androidx.camera.camera2.e.v.run
  at android.os.Handler.handleCallback (Handler.java:815)
  at android.os.Handler.dispatchMessage (Handler.java:104)
  at android.os.Looper.loop (Looper.java:207)
  at android.app.ActivityThread.main (ActivityThread.java:5777)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:789)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:679)
```

Also, not sure yet if this one is related to this specific issue, but the CameraPage also gives this error:

```
com.facebook.react.bridge.JSApplicationIllegalArgumentException: 
  at com.facebook.react.animated.l.g (Unknown Source:79)
  at com.facebook.react.animated.NativeAnimatedModule$f.a (Unknown Source:4)
  at com.facebook.react.animated.NativeAnimatedModule.executeAllOperations (Unknown Source:31)
  at com.facebook.react.animated.NativeAnimatedModule.access$400 (Unknown Source)
  at com.facebook.react.animated.NativeAnimatedModule$p.execute (Unknown Source:8)
  at com.facebook.react.uimanager.z0$t.m (Unknown Source:8)
  at com.facebook.react.uimanager.z0$a.run (Unknown Source:135)
  at com.facebook.react.uimanager.z0.U (Unknown Source:53)
  at com.facebook.react.uimanager.z0.s (Unknown Source)
  at com.facebook.react.uimanager.z0$j.doFrameGuarded (Unknown Source:31)
  at com.facebook.react.uimanager.f.doFrame (Unknown Source)
  at com.facebook.react.modules.core.g$d.doFrame (Unknown Source:46)
  at com.facebook.react.modules.core.a$a$a.doFrame (Unknown Source:2)
  at android.view.Choreographer$CallbackRecord.run (Choreographer.java:1106)
  at android.view.Choreographer.doCallbacks (Choreographer.java:866)
  at android.view.Choreographer.doFrame (Choreographer.java:792)
  at android.view.Choreographer$FrameDisplayEventReceiver.run (Choreographer.java:1092)
  at android.os.Handler.handleCallback (Handler.java:938)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at android.os.Looper.loopOnce (Looper.java:226)
  at android.os.Looper.loop (Looper.java:313)
  at android.app.ActivityThread.main (ActivityThread.java:8680)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:567)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1135)
```

Does that help in any way? :/

mrousavy, 2022-04-19:
Unfortunately not, no.

The logs you shared are stripped/minified, you need to either give me adb logs or symbolicated stacktraces (e.g. when not using Proguard).

The other error is an animation/reanimated error.