margelo/react-native-vision-camera #1005: 🐛  Issue with frame process. Works in debug but not in release.

sanketkheni01, 2022-04-18:
### What were you trying to do?

Building release build. The QR scanner works without any issue in debug mode but not in release mode.

### Reproduceable Code

```tsx
export default function ScanScreen({navigation}: {navigation: Navigation}) {
  // * States
  const [QrData, setQrData] = useState<string | undefined>('')
  const [isVerified, setIsVerified] = useState(false)

  // * Hooks
  const [frameProcessor, barcodes] = useScanBarcodes([BarcodeFormat.QR_CODE], {
    checkInverted: false,
  })

  // * Effects
  // ? When camera scan and get somethings
  useEffect(() => {
    if (barcodes.length > 0) {
      setQrData(barcodes[0]?.displayValue)
    }
  }, [barcodes])

  // ? Verify on getting a new qr code
  useEffect(() => {
    handleQrData()
  }, [QrData])

  // * Functions
  const handleQrData = async () => {
    if (QrData) {
      const deviceData: ConnectionDataDesktop = JSON.parse(QrData)

      const result: boolean = await verify(deviceData, setSocket)
      if (result) {
        // REF **storage** "desktopData" => all the desktop data ConnectionDataDesktop type
        await AsyncStorage.setItem('desktopData', QrData)
        setIsVerified(true)
      } else {
        setQrData('')
      }
    }
  }

  return (
    <View style={tw`w-full h-full items-center justify-center bg-white`}>
      <Text style={tw`font-soraM mb-5 text-xl text-black`}>
        Scan to connect
      </Text>

      {/* Main qr code camera component */}
      <View style={tw`w-72 h-72`}>
        <QrScanner
          isLoading={QrData ? true : false}
          frameProcessor={frameProcessor}
        />
      </View>
    </View>
  )
}
```

### What happened instead?

Camera blinks.
https://user-images.githubusercontent.com/68942390/163852482-f4bf5f30-9079-41df-b2f1-96c117aeb079.mp4

### Relevant log output

```shell
2022-04-18 23:24:03.462 28827-28858/? E/unknown:SoftAssertions: Unhandled SoftException
    com.facebook.react.bridge.AssertionException: Expected to run on UI thread!
        at com.facebook.react.bridge.SoftAssertions.assertCondition(SoftAssertions.java:37)
        at com.facebook.react.bridge.UiThreadUtil.assertOnUiThread(UiThreadUtil.java:26)
        at com.facebook.react.uimanager.UIManagerModule.resolveView(UIManagerModule.java:957)
        at com.mrousavy.camera.frameprocessor.FrameProcessorRuntimeManager.findCameraViewById(FrameProcessorRuntimeManager.kt:65)
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27)
        at android.os.Looper.loopOnce(Looper.java:201)
        at android.os.Looper.loop(Looper.java:288)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:228)
        at java.lang.Thread.run(Thread.java:920)
2022-04-18 23:24:03.482 28827-28906/? E/BufferQueueProducer: [SurfaceView[com.justshare/com.justshare.MainActivity]#1(BLAST Consumer)1](id:709b00000003,api:4,p:1080,c:28827) queueBuffer: BufferQueue has been abandoned
2022-04-18 23:24:03.482 1080-18570/? E/Surface: queueBuffer: error queuing buffer, -19
2022-04-18 23:24:03.482 1080-18570/? E/Camera3-OutputStream: returnBufferCheckedLocked: Stream 0: Error queueing buffer to native window: No such device (-19)
2022-04-18 23:24:03.515 28827-28861/? E/BufferQueueProducer: [SurfaceView[com.justshare/com.justshare.MainActivity]#1(BLAST Consumer)1](id:709b00000003,api:4,p:1080,c:28827) queueBuffer: BufferQueue has been abandoned
2022-04-18 23:24:03.515 1080-18570/? E/Surface: queueBuffer: error queuing buffer, -19
2022-04-18 23:24:03.515 1080-18570/? E/Camera3-OutputStream: returnBufferCheckedLocked: Stream 0: Error queueing buffer to native window: No such device (-19)
2022-04-18 23:24:03.536 28827-28906/? E/BufferQueueProducer: [SurfaceView[com.justshare/com.justshare.MainActivity]#1(BLAST Consumer)1](id:709b00000003,api:4,p:1080,c:28827) cancelBuffer: BufferQueue has been abandoned
2022-04-18 23:24:03.538 28827-28906/? E/BufferQueueProducer: [SurfaceView[com.justshare/com.justshare.MainActivity]#1(BLAST Consumer)1](id:709b00000003,api:4,p:1080,c:28827) cancelBuffer: BufferQueue has been abandoned
```

### Device

Android 12

### VisionCamera Version

2.13.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-04-19:
Hey, which RN version are you using?

sanketkheni01, 2022-04-19:
Sorry for that after a very long debugging. I found that issue wasn't with the vision camera it was some other code that was making the camera flicker.

closed 2022-04-19 by sanketkheni01

eramudeep, 2022-04-29:
> ```tsx
> useScanBarcodes
> ```

hey can you please share your code .?
i am also looking to implement something like this.

sanketkheni01, 2022-04-29:
> > ```tsx
> > useScanBarcodes
> > ```
> 
> hey can you please share your code .?
> i am also looking to implement something like this. 

https://github.com/rodgomesc/vision-camera-code-scanner

eramudeep, 2022-04-29:
> > > ```tsx
> > > useScanBarcodes
> > > ```
> > 
> > 
> > hey can you please share your code .?
> > i am also looking to implement something like this.
> 
> https://github.com/rodgomesc/vision-camera-code-scanner

worked like a charm. Thank you !!!!

rodgomesc, 2022-05-17:
i really dont understand how some devs can build my project sucessfully but i can't, @mrousavy  i understand that's not your business, but if you can put me on the right direction on this https://github.com/rodgomesc/vision-camera-code-scanner/issues/59 i really appreciate, this error doesn't make much sense, I already went back commits where I was sure the project worked and even then I always end up getting this error when running the project, tried downgrade and upgrade reanimated dependency as well as vision camera version

mrousavy, 2022-05-24:
Honestly I have no idea. Those errors are reanimated errors (I think) and they are hard to debug. Maybe this will be fixed by migrating to Fabric (both REA and VisionCamera)

sergeymild, 2022-12-06:
Have the same issue:

```
Unhandled SoftException
                 com.facebook.react.bridge.AssertionException: Expected to run on UI thread!
                 	at com.facebook.react.bridge.SoftAssertions.assertCondition(SoftAssertions.java:37)
                 	at com.facebook.react.bridge.UiThreadUtil.assertOnUiThread(UiThreadUtil.java:26)
                 	at com.facebook.react.uimanager.UIManagerModule.resolveView(UIManagerModule.java:957)
                 	at com.mrousavy.camera.CameraViewModule.findCameraView(CameraViewModule.kt:87)
                 	at com.mrousavy.camera.CameraViewModule.access$findCameraView(CameraViewModule.kt:34)
                 	at com.mrousavy.camera.CameraViewModule$takePhoto$1.invokeSuspend(CameraViewModule.kt:96)
                 	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                 	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
                 	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
                 	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
                 	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
                 	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
```

adamdavareln, 2022-12-09:
Hi @sanketkheni01 , what is your solution to fix the flickering camera? I experiencing the same issue but not the same implementation. I use frame processor to detect the face using this library ( [vision-camera-face-detector](https://github.com/rodgomesc/vision-camera-face-detector) and the camera is flickering. Please help me 🥲

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

Try upgrading to V3 if you can, it has a much more stable yet flexible foundation.

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.