margelo/react-native-vision-camera #999: 🐛 React native vison camera build stopped at 92%

badalchangba, 2022-04-18:
### What were you trying to do?

[image]
[image]
[image]

### Reproduceable Code

```tsx
I have java 11 and running the react app with proper react-native vison configuration and running the app.

npm run android
```
but my build configuration stops at 92%. What might be the cause for this issue?
```

### What happened instead?

My build configuration get stuck at 92%

### Device

Honor 9N

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

syedtehrimabbas, 2022-04-18:
Having the same issue on MacBook M1 Pro with device android 12.
[image]

mrousavy, 2022-04-19:
Looks like you don't have the NDK installed and it is trying to install it, which just takes a very long time apparently.

This is mentioned [in the Troubleshooting guide](https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting#android).

Install the NDK separately, clean cache maybe restart your PC and then try again

closed 2022-04-19 by mrousavy

EKOzkan, 2022-07-31:
If anybody still experiences this same issue that is caused by ndk setup after the doing the steps in the [Troubleshooting guide](https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting/#android),  try installing the **exact ndk version** that you see in the terminal text via android studio. 

Here is a little guide for this:

Check the version first

[image]

Go to Android Studio - Sdk Manager, then get to the Sdk tools tab. Then toggle the "Show Package Details" option. 

[image]

And download the **exact version** that you see in your terminal.

[image]

And you should be good to go. If you encounter any errors based on ndk, you can try this too:

Go to your projects android folder, create a file called "local.properties", and inside that file enter your sdk location like this:

`sdk.dir=C:\\Users\\!!!USERNAMEHERE!!!\\AppData\\Local\\Android\\Sdk`