margelo/react-native-vision-camera #1007: 🐛 Error: Cannot access realm that has been closed.

JayadeepSyngenta, 2022-04-20:
### What were you trying to do?

[image]
.

Without using "react-native-vision-camera" dependency the App was build and working fine but after installing the "react-native-vision-camera" dependency the App was crashing with the Error: "Cannot access realm that has been closed" at the place where the "realm" instance was used

### Reproduceable Code

```tsx
Using below dependencies in my project
"react-native-vision-camera": "2.13.0",
"realm": "^10.4.1",
```

### What happened instead?

The app crashes with Error: Cannot access realm that has been closed.

### Relevant log output

```shell
ReactNativeJS: Error: Cannot access realm that has been closed.
    
    This error is located at:
        in OtpScreen (created by Connect(OtpScreen))
        in Connect(OtpScreen) (at SceneView.tsx:130)
        in StaticContainer
        in EnsureSingleNavigator (at SceneView.tsx:122)
        in SceneView (at useDescriptors.tsx:181)
        in RCTView (at View.js:34)
        in View (at CardContainer.tsx:245)
        in RCTView (at View.js:34)
        in View (at CardContainer.tsx:244)
        in RCTView (at View.js:34)
        in View (at CardSheet.tsx:33)
        in CardSheet (at Card.tsx:573)
        in RCTView (at View.js:34)
        in View (at createAnimatedComponent.js:217)
        in AnimatedComponent (at createAnimatedComponent.js:278)
        in AnimatedComponentWrapper (at Card.tsx:555)
        in PanGestureHandler (at GestureHandlerNative.tsx:13)
        in PanGestureHandler (at Card.tsx:549)
        in RCTView (at View.js:34)
        in View (at createAnimatedComponent.js:217)
        in AnimatedComponent (at createAnimatedComponent.js:278)
        in AnimatedComponentWrapper (at Card.tsx:544)
        in RCTView (at View.js:34)
        in View (at Card.tsx:538)
        in Card (at CardContainer.tsx:206)
        in CardContainer (at CardStack.tsx:623)
        in RNSScreen (at createAnimatedComponent.js:217)
        in AnimatedComponent (at createAnimatedComponent.js:278)
        in AnimatedComponentWrapper (at src/index.native.tsx:281)
        in MaybeFreeze (at src/index.native.tsx:280)
        in Screen (at Screens.tsx:74)
        in MaybeScreen (at CardStack.tsx:616)
        in RNSScreenContainer (at src/index.native.tsx:351)
        in ScreenContainer (at Screens.tsx:50)
        in MaybeScreenContainer (at CardStack.tsx:498)
        in CardStack (at StackView.tsx:462)
        in KeyboardManager (at StackView.tsx:458)
        in RNCSafeAreaProvider (at SafeAreaContext.tsx:76)
        in SafeAreaProvider (at SafeAreaProviderCompat.tsx:42)
        in SafeAreaProviderCompat (at StackView.tsx:455)
        in GestureHandlerRootView (at GestureHandlerRootView.android.tsx:26)
        in GestureHandlerRootView (at StackView.tsx:454)
        in StackView (at createStackNavigator.tsx:87)
        in StackNavigator (at RootNavigation.tsx:18)
        in EnsureSingleNavigator (at BaseNavigationContainer.tsx:411)
        in BaseNavigationContainer (at NavigationContainer.tsx:91)
        in ThemeProvider (at NavigationContainer.tsx:90)
        in NavigationContainer (at RootNavigation.tsx:17)
        in RootNavigation (at App.tsx:170)
        in App (created by Connect(App))
        in Connect(App) (at mobile-react-native-grower/index.js:17)
        in Provider (at mobile-react-native-grower/index.js:16)
        in MyApp (at renderApplication.js:47)
        in RCTView (at View.js:34)
        in View (at AppContainer.js:107)
        in RCTView (at View.js:34)
        in View (at AppContainer.js:134)
        in AppContainer (at renderApplication.js:40)
```

### Device

Samsung A51 (android versin - 11)

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-04-20:
Closing because no reproduceable sample was provided. VisionCamera  doesn't really have anything to do with Realm, and without a test project I cannot reproduce this.

closed 2022-04-20 by mrousavy

JayadeepSyngenta, 2022-04-21:
@mrousavy if only 'realm' lib was used it was working fine but when the 'react-native-vision-camera' lib was added the app was crashing at place where realm instance was used.

Please download the Reproduceable Code sample from the below link https://github.com/jayadeep7/ReactNativeVisionCameraPoc

olive49, 2022-07-18:
@JayadeepSyngenta were you able to solve this issue?

JayadeepSyngenta, 2022-07-21:
> @JayadeepSyngenta were you able to solve this issue?

Updating the react-native version and realm lib version the issue got resolved