margelo/react-native-vision-camera #1008: 🐛  App crashes when switching front/back camera while recording a video.

usaidather, 2022-04-20:
### What were you trying to do?

While recording a video when I switches the camera from front to back or vice versa quickly and doing it many times it crashes.

### Reproduceable Code

```tsx
import React, { useEffect, useCallback, useRef, useState } from "react";
import {
  StyleSheet,
  View,
  TouchableOpacity,
  Linking,
  Platform,
  AppState,
  Modal,
} from "react-native";
import { Button, ButtonWithImage, Text } from "../components";
import { useIsFocused } from "@react-navigation/core";
import { ColorConst, ImageConst } from "../constants";
import { Camera, useCameraDevices } from "react-native-vision-camera";
import { SizeClass } from "../utils/AppTheme";
import { FONT_FAMILY, APP_FONTS } from "../utils/FontsUtils";
import Timer from "./Timer";
import { useDispatch } from "react-redux";
import { timerReset } from "../redux/Timer/TimerSlice";
import { SafeAreaView } from "react-native-safe-area-context";
import { deleteFile } from "../utils/Utility";

export default function VidiAskCamera(props) {
  const {
    showBackCamera,
    setShowBackCamera,
    isRecording,
    setIsRecording,
    recordedVideo,
    setRecordedVideo,
    navigation,
    setVideoPath,
    replyTo,
  } = props;

  const devices =
    showBackCamera === true
      ? useCameraDevices("wide-angle-camera")
      : useCameraDevices();

  const isFocused = useIsFocused();
  const cameraRef = useRef();
  const [cameraPermission, setCameraPermission] = useState();
  const [micPermission, setMicPermission] = useState();
  const [flashMessageModalVisible, setFlashMessageModalVisible] =
    useState(false);

  const [onCameraFlip, setOnCameraFlip] = useState(false);

  const device = showBackCamera === true ? devices.back : devices.front;

  const appStateVar = useRef(AppState.currentState);

  const dispatch = useDispatch();

  const reqPer = async () => {
    try {
      const newCameraPermission = await Camera.requestCameraPermission();
      const newMicrophonePermission =
        await Camera.requestMicrophonePermission();
      setCameraPermission(newCameraPermission);
      setMicPermission(newMicrophonePermission);
    } catch (error) {}
[... 392 lines cut ...]
  let timestamp = CMSyncConvertTime(CMSampleBufferGetPresentationTimeStamp(sampleBuffer),
                                          from: audioCaptureSession.masterClock!,
                                          to: captureSession.masterClock!)
which is line no 204 of file CameraView+RecordVideo inside the function captureOutput

### Relevant log output

```shell
VisionCamera/CameraView+RecordVideo.swift:204: Fatal error: Unexpectedly found nil while unwrapping an Optional value
```

### Device

iPhone 12

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-04-20:
Hi, thanks for the report. Is `audioCaptureSession.masterClock` nil or is it `captureSession.masterClock`?

usaidather, 2022-04-20:
Hi mrousavy thanks for quick response. "to: captureSession.masterClock!" is nil

jacquesikot, 2022-07-21:
@usaidather Hello, I am having the same issue, did you find a solution yet?

meetkkvelu, 2022-10-21:
Hello, I am also having the same issue

mrousavy, 2022-10-24:
Does https://github.com/mrousavy/react-native-vision-camera/pull/1302 fix your issue?

ridvanaltun, 2023-01-18:
> Does #1302 fix your issue?

I was have the same issue, the PR works

closed 2023-06-15 by mrousavy

meetkkvelu, 2023-09-15:
i am also facing same issue