margelo/react-native-vision-camera #1011: ✨ How can I save the video recording when the app state changes to "background" or "inactive".

hblab-hungpv, 2022-04-21:
### What feature or enhancement are you suggesting?

 I want to save the video recording when the app state changes to "background" or "inactive".

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

How can I save the video I'm recording before the app state changes to "background".
Currently while recording a video, I press the 'HOME' button on the device, which immediately lands on: onRecordingError. Can I handle the recording of the recorded video before it falls on the onRecordingError?
Please help me. Thanks.
camera.current.startRecording({
         onRecordingError: error => {
           console.error('Recording failed!', error);
           onStoppedRecording();
         },
         onRecordingFinished: video => {
           console.log(`Recording successfully finished! ${video.path}`);

           onMediaCaptured(video, 'video');
           onStoppedRecording();
         },
       });

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

usaidather, 2022-04-27:
I am also looking for this issue... Is there any workaround.

rahul2992, 2022-08-31:
+1 for this feature. I have made it work on iOS by using a custom hook to detect and stop the video recording when the app goes in background. However, on android the Camera errors without giving the video file.

OCCCAS, 2023-01-14:
Sorry, are you fixed this problem? When I am trying to stop video, program throws error: `[capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames`
Please help!

Sid-Turner-Ellis, 2023-02-22:
Think this is a bug rather than a feature request? the [docs](https://mrousavy.com/react-native-vision-camera/docs/guides/lifecycle/#the-isactive-prop) suggest that it is capable of working in the background, also worth noting that this occurs when you take a call too.

So a few issues here:

1. Background recording not working
    
    After looking into CameraX and the Android SDK it would seem to get background recording working reliably (to not get killed by the system) you would need to bind the cameras lifecycle to the lifecycle of a foreground service - I got this working in a native app and it worked fine. RNVC looks to be using a custom lifecycle but its not bound to a foreground service so runs the risk of being killed.
    
2. RNVC lifecycle is following the lifecycle of the host
    
    When the main activity goes into the background, this code runs which sets the lifecycle to created which seems to cause CameraX to reinitialise, if you comment out the logic here then the recording will not get killed straight away (though it will eventually)
    
    ```
    override fun onHostPause() {
      hostLifecycleState = Lifecycle.State.CREATED
      updateLifecycleState()
    }
    ```
    
3. When the app comes back into the foreground, the use cases are unbound and the session is re-configured and starts again. I can see the reason for it links an issue that is was marked as fixed in October, I updated the build.gradle file to use the latest versions (1.3.0-alpha03) and commented out the workaround line and it seems that upon returning to the app the preview doesn’t show (even though the camera still records)
    
    ```
    override fun onHostResume() {
      hostLifecycleState = Lifecycle.State.RESUMED
      updateLifecycleState()
      // workaround for https://issuetracker.google.com/issues/147354615, preview must be bound on resume
      **update(propsThatRequireSessionReconfiguration)**
    }
    ```

As time is tight in my case i'm going to revisit this straight after I add a workaround which will probably be to make a patch around the file names and then stitching them together.

@mrousavy off the top of your head, do you know of anyway to introduce background recording on Android?

Sid-Turner-Ellis, 2023-02-22:
Have just seen there is a similar query here: https://github.com/mrousavy/react-native-vision-camera/issues/1408

@usaidather @rahul2992 @OCCCAS @hblab-hungpv did you guys have any luck with this?

Sid-Turner-Ellis, 2023-02-22:
@mrousavy happy to work on this as soon as i've implemented the workaround for the project i'm working on, what specifically do you want a PR to address? (foreground service, preventing onStop error or encapsulating the logic for a workaround?)

ratz6, 2023-03-01:
same issue on android, works on iOS

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy