margelo/react-native-vision-camera #1012: 🐛 Task :react-native-vision-camera:compileDebugKotlin FAILED

bekatd, 2022-04-22:
### What were you trying to do?

I am trying to install this library

### What happened instead?

Build fails at this step: `> Task :react-native-vision-camera:compileDebugKotlin FAILED`

### Relevant log output

```shell
> Task :react-native-vision-camera:compileDebugKotlin FAILED
w: Runtime JAR files in the classpath should have the same version. These files were found in the classpath:
    user/.gradle/caches/transforms-2/files-2.1/664f78d5defbcdceed6cc6b91334957c/jetified-kotlin-stdlib-jdk8-1.5.30.jar (version 1.5)
    user/.gradle/caches/transforms-2/files-2.1/98f292b47b7ae2a67fdb09d889c153dc/jetified-kotlin-stdlib-jdk7-1.5.30.jar (version 1.5)
    user/.gradle/caches/transforms-2/files-2.1/2482ed84b3d2e1c78620a60749fa204b/jetified-kotlin-stdlib-1.6.10.jar (version 1.6)     
    user/.gradle/caches/transforms-2/files-2.1/a1dc49042920562f5109600bb5b5b2b0/jetified-kotlin-stdlib-common-1.6.10.jar (version 1.6)     
w: Some runtime JAR files in the classpath have an incompatible version. Consider removing them from the classpath

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.7.1/userguide/command_line_interface.html#sec:command_line_warnings
581 actionable tasks: 8 executed, 573 up-to-date
e: \node_modules\react-native-vision-camera\android\src\main\java\com\mrousavy\camera\frameprocessor\FrameProcessorRuntimeManager.kt: (65, 76): Unresolved reference: resolveView

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> Compilation error. See log for more details

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 47s
```

### Device

Xiaomi mi A2 lite (real device)

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bekatd, 2022-04-22:
I tried everything from A to Z found on internet (searching for almost 30 hours). Hope there is some solution :|

mrousavy, 2022-04-22:
Hi, you didn't post an error message. Only warnings from the log are here (`w:`).
Please post full output.

closed 2022-04-22 by mrousavy

bekatd, 2022-04-22:
[image]

how is this only warning

bekatd, 2022-04-22:
Please tell me, what type of log can I give you additionally?

mrousavy, 2022-04-22:
Ah my bad I just didn't see the error correctly. You're right, this is available only on newer RN versions

RafaelPena31, 2022-04-22:
I have this problem too, do we have any updates?

bekatd, 2022-04-22:
See my comment above about downgrade

vgorte, 2022-04-24:
This seems to be related to #957.

raulburigo, 2022-04-27:
The downgrade did not work here.

"react-native": "0.63.2"
"react-native-vision-camera": "2.13.0"

ddmarinova, 2022-04-29:
I get the same issue. Downgrading to "react-native-vision-camera": "2.13.0" doesn't work for me. I ended up downgrading to 1.0.11, which seems to be working.

"react-native": "0.64.1"
"react-native-vision-camera": "^1.0.11"

dhavallogistic, 2022-05-03:
> The downgrade did not work here.
> 
> "react-native": "0.63.2" "react-native-vision-camera": "2.13.0"

Same here.. Downgrade is also not working for me.
I had installed version **_"2.11.2"_**
After changing and tried with version **"2.13.2"**, **"2.13.1"** & **"2.13.2"** facing above error.

And even after changed to old version i.e. "2.11.2", 
node_modules packages are updated so that error show continue while building Android code.

bekatd, 2022-05-03:
Try to run gradlew clean and delete node modules. Also if you have old expo unimodules installation, try to migrate to new expo modules

bekatd, 2022-05-03:
Also please post screenshot from android studio, what went wrong exactly

dhavallogistic, 2022-05-03:
> Try to run gradlew clean and delete node modules. Also if you have old expo unimodules installation, try to migrate to new expo modules

Yes, did all of above.

- Android Clean gradle
- Rebuild gradle
- Remove all node packages and install from scratch
- Android studio Invalidate cache & Restart

But even after all these, got this error 

[image]

bekatd, 2022-05-03:
But error you posted can't happen in case of 2.13.0. You are trying to install 2.13.2 right?

bekatd, 2022-05-03:
I had exact same error and resolved by downgrading to 2.13.0

dhavallogistic, 2022-05-03:
> But error you posted can't happen in case of 2.13.0. You are trying to install 2.13.2 right?

No, I tested with version 2.13.0

[image]

dhavallogistic, 2022-05-03:
> I had exact same error and resolved by downgrading to 2.13.0

Yes I saw your post and solution you gave. And I also tried to implement it and after all that trial and error I got this result with all versions.
Only working version is 2.11.2 and that is also with old node packages. Otherwise no luck :(

bekatd, 2022-05-03:
But simply v2.13.0 does not have the code you are showing on screenshot of error

dhavallogistic, 2022-05-03:
You mean to say, the file I am getting an error called `FrameProcessorRuntimeManage.kt` is not included in v2.13.0?

bekatd, 2022-05-03:
Yes. You are somehow installing latest version instead of 2.13.0

Here is the link to source of that file https://github.com/mrousavy/react-native-vision-camera/blob/65d4d46f6ab4e04dca15c15b8cf9c07a17f97d72/android/src/main/java/com/mrousavy/camera/frameprocessor/FrameProcessorRuntimeManager.kt

bekatd, 2022-05-03:
Just delete node_modules, edit lib version in package.json to 2.13.0, run yarn or npm, run gradlew clean and thats it

ddmarinova, 2022-05-04:
I have done all the steps previously mentioned (started from scratch):
 - node_modules delete
 - gradle clean
 - rebuild gradle
 
FrameProcessorRuntimeManager.kt: (65, 76): Unresolved reference: resolveView persists on every major version, except for the first one. Downgrading to 2.11.2 doesn't work for me.

mrousavy, 2022-05-05:
That is a RN version error, are you on 0.67?

Venglevskyi, 2022-05-16:
Same problem. Have any suggestions??

dhavallogistic, 2022-05-16:
> Same problem. Have any suggestions??

In my case below solutions is working.. 
1. Remove all node files
2. `npm i`
3. Try to install with particular version like.. `npm i react-native-vision-camera@2.13.0`

MuhammadAbdullah54321, 2022-05-25:
I was facing the same issue.
Banged my head for hours and then found this silly issue 
I was using version `2.13.3` and now I have downgraded to `2.13.0` and its working

simply do `yarn add react-native-vision-camera@2.13.0`

MuhammadAbdullah54321, 2022-05-27:
Downgrading to `2.13.0` works fine for both android and iOS but some features like zoom and auto light adjustment wont work on iOS which are actually available in 2.13.3

aniruddhashevle, 2022-07-13:
Downgrading to version 2.13.0 works fine.

SpectreWulf, 2022-08-23:
Facing same failed condition!

Tried lower versions but that didn't help.

Android
React Native: 0.61.5
Vision Camera: 2.14.1

MuhammadAbdullah54321, 2022-08-23:
> Facing same failed condition!
> 
> Tried lower versions but that didn't help.
> 
> Android React Native: 0.61.5 Vision Camera: 2.14.1

May be you should try to upgrade React Native version

SpectreWulf, 2022-08-23:
> 

Which React Native version are you using?

SpectreWulf, 2022-08-23:
> May be you should try to upgrade React Native version

Which React Native version are you using?

MuhammadAbdullah54321, 2022-08-23:
I am using React Native 0.63.4 and vision camera version 2.13.0 ... still `takeSnapshot` for android doesn't work but `takePhoto` works for both

MuhammadAbdullah54321, 2022-08-23:
> > May be you should try to upgrade React Native version
> 
> Which React Native version are you using?

I am using React Native 0.63.4 and vision camera version 2.13.0 ... still `takeSnapshot` for android doesn't work but `takePhoto` works for both

SpectreWulf, 2022-08-23:
> > > May be you should try to upgrade React Native version
> > 
> > 
> > Which React Native version are you using?
> 
> I am using React Native 0.63.4 and vision camera version 2.13.0 ... still `takeSnapshot` for android doesn't work but `takePhoto` works for both

Thanks I will try to upgrade but React native upgrade is a lot of pain especially from an older version for an existing app! 😥

kofkgoing, 2022-11-03:
[image]

kofkgoing, 2022-11-03:
I found a solution.
resolveView method was inserted at RN 0.66.
- https://github.com/facebook/react-native/commit/de255528e0b65980d5531c477fb2b32bbbcd9b75

minor RN can't use latest vision camera.

To fix this.
open FrameProcessorRuntimeManger.kt and edit
```
val view: CameraView? = ctx?.currentActivity?.findViewById<CameraView>(viewId)
//    val view = if (ctx != null) UIManagerHelper.getUIManager(ctx, viewId)?.resolveView(viewId) as CameraView? else null
```

open CameraViewModule.kt and edit
```
val view: CameraView? = ctx?.currentActivity?.findViewById<CameraView>(viewId)
//    val view = if (ctx != null) UIManagerHelper.getUIManager(ctx, viewId)?.resolveView(viewId) as CameraView? else null
```

I can't use 0.66, my project use 0.64

inkCrazy, 2022-11-22:
FAILED
react native 0.63.5
Vision Camera: 2.13.0

nottalha, 2023-02-22:
This worked for me, 
IMPORTANT: "react-native-gesture-handler:" "^2.9.0" was the problem, changed to to "1.10.3".

    "react": "17.0.1",
    "react-native": "0.64.4",
    "react-native-gesture-handler": "1.10.3",
    "react-native-vision-camera": "^2.13.0"

AjayParambath, 2023-05-11:
> Downgrading to 2.13.0 fixed it for me. I don't know why.
> 
> If it is related to the RN version, I think it should be mentioned in the docs
> 
> I am using RN 0.64.1

     "react": "17.0.2",
    "react-native": "^0.64.2",
    gradle version: 
    com.android.tools.build:gradle:4.2.0
gradle-6.7.1

"react-native-vision-camera": "2.13.0"

cleared node modules and package-lock.json
ran this command in terminal : cd android ; ./gradlew clean; cd .. ; npm i;

This worked

bernborgess, 2023-07-28:
Any updates on this? In the sense that the current expo sdk and react native version will support react-native-vision-camera properly?
expo: 49.0.5
react-native: 0.72.3
react-native-vision-camera: 2.15.4

mkurczewski, 2023-08-04:
For me downgrading Kotlin worked well

Just change the `kotlinVersion` in `android/build.gradle` to 1.7.21 (I didn't check other versions):

```
buildscript {
  ext {
    ...
    kotlinVersion = '1.7.21'
    ...
  }
  ...
}
```

kundakcii, 2023-08-30:
> 

what is your RN version

OPEN-9, 2023-10-11:
https://github.com/mrousavy/react-native-vision-camera/issues/1977

2.16.2 This problem occurs

Georg7, 2024-05-13:
I have the same issue: 
```
 "react-native": "0.74.1",
"react-native-vision-camera": "^4.0.3",
"expo": "51.0.4",
```
Expo managed build results in: ```Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.``` Works fine on iOS. 

Tried most of the suggestions in this thread but nothing seems to work - all resulting in the same error.

losheredos, 2024-05-15:
@Georg7 I have the same issue, tried changing versions as suggested in this [link](https://github.com/mrousavy/react-native-vision-camera/issues/1012) but nothing helps unfortunately.

> Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> > A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
>    > Compilation error. See log for more details

```
"react-native": "0.74.0",
"react-native-vision-camera": "^4.0.4",
```

> android/build.gradle

```
buildscript {
    ext {
        buildToolsVersion = "34.0.0"
        minSdkVersion = 23
        compileSdkVersion = 34
        targetSdkVersion = 34
        ndkVersion = "26.1.10909125"
        kotlinVersion = "1.9.24"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle")
        classpath("com.facebook.react:react-native-gradle-plugin")
        classpath 'com.google.gms:google-services:4.4.1'
        classpath 'com.google.firebase:perf-plugin:1.4.2'
        classpath 'com.google.firebase:firebase-crashlytics-gradle:2.9.9'
        classpath("org.jetbrains.kotlin:kotlin-gradle-plugin")
    }
}

apply plugin: "com.facebook.react.rootproject"

```

Daniel-i-m, 2024-06-02:
> @Georg7 I have the same issue, tried changing versions as suggested in this [link](https://github.com/mrousavy/react-native-vision-camera/issues/1012) but nothing helps unfortunately.
>
>
> ``` [...]

same here any solution please, tried downgrading react-native react-native-vision camera kotlin version but nothing works. same error keeps comming.. thinking of using deprecated react-native-camera now