margelo/react-native-vision-camera #1014: ❓ Do I need to include mlkit manually to use frameprocessors?

bekatd, 2022-04-23:
### Question

Android app crashes immediately upon launch with following crash log:

```
04-23 10:27:39.385  2657  2657 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 2657 (init), pid 2657 (init)
04-23 10:27:39.630  2657  2657 F libc    : crash_dump helper failed to exec
04-23 10:32:20.155 11221 11356 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 11356 (glean.dispatche), pid 11221 (AndroidUI)
04-23 10:32:20.683 11549 11549 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
04-23 10:32:20.683 11549 11549 F DEBUG   : Build fingerprint: 'xiaomi/daisy/daisy_sprout:10/QKQ1.191002.002/V11.0.21.0.QDLMIXM:user/release-keys'
04-23 10:32:20.683 11549 11549 F DEBUG   : Revision: '0'
04-23 10:32:20.683 11549 11549 F DEBUG   : ABI: 'arm64'
04-23 10:32:20.684 11549 11549 F DEBUG   : Timestamp: 2022-04-23 10:32:20+0400
04-23 10:32:20.684 11549 11549 F DEBUG   : pid: 11221, tid: 11356, name: glean.dispatche  >>> org.torproject.torbrowser <<<
04-23 10:32:20.684 11549 11549 F DEBUG   : uid: 10154
04-23 10:32:20.684 11549 11549 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
04-23 10:32:20.684 11549 11549 F DEBUG   : Cause: null pointer dereference
04-23 10:32:20.684 11549 11549 F DEBUG   :     x0  00000074236dc50a  x1  000000000000008d  x2  00000074236dc302  x3  00000074236dc680
04-23 10:32:20.684 11549 11549 F DEBUG   :     x4  00000074236dc2c8  x5  00000074236dc70a  x6  0000000000000000  x7  0000000000000000
04-23 10:32:20.684 11549 11549 F DEBUG   :     x8  0000000000000000  x9  000000744ccb4990  x10 0000000000000000  x11 0000000000000000
04-23 10:32:20.684 11549 11549 F DEBUG   :     x12 0000000000000000  x13 0000000000000000  x14 0000000000000100  x15 0000000000000001
04-23 10:32:20.684 11549 11549 F DEBUG   :     x16 0000007439298df8  x17 000000753bcab380  x18 000000741d466000  x19 0000000000000047
04-23 10:32:20.684 11549 11549 F DEBUG   :     x20 0000007437d88af2  x21 000000000000008d  x22 00000074236dc300  x23 00000074236dc508
04-23 10:32:20.684 11549 11549 F DEBUG   :     x24 00000074393ab8e8  x25 0000007437d2914c  x26 0000000000000006  x27 0000000000000001
04-23 10:32:20.684 11549 11549 F DEBUG   :     x28 0000007437d29164  x29 0000007437d29164
04-23 10:32:20.684 11549 11549 F DEBUG   :     sp  00000074236dc2f0  lr  0000007435e31c44  pc  00000074368ef1ac
04-23 10:32:20.738 11549 11549 F DEBUG   :
04-23 10:32:20.738 11549 11549 F DEBUG   : backtrace:
04-23 10:32:20.738 11549 11549 F DEBUG   :       #00 pc 00000000048e11ac  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.738 11549 11549 F DEBUG   :       #01 pc 0000000003e23c40  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.738 11549 11549 F DEBUG   :       #02 pc 0000000003e23b7c  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.739 11549 11549 F DEBUG   :       #03 pc 0000000004004944  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.739 11549 11549 F DEBUG   :       #04 pc 0000000004007260  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.739 11549 11549 F DEBUG   :       #05 pc 00000000040071f0  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.739 11549 11549 F DEBUG   :       #06 pc 0000000004004488  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.739 11549 11549 F DEBUG   :       #07 pc 00000000009c747c  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.739 11549 11549 F DEBUG   :       #08 pc 00000000009c76d4  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.739 11549 11549 F DEBUG   :       #09 pc 0000000003c387a0  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.739 11549 11549 F DEBUG   :       #10 pc 0000000003d3b1ec  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.739 11549 11549 F DEBUG   :       #11 pc 0000000003d39a00  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.739 11549 11549 F DEBUG   :       #12 pc 0000000004007af4  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:32:20.739 11549 11549 F DEBUG   :       #13 pc 00000000000e6890  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+36) (BuildId: cf739dbc84bcc78f7a1500721bfb3758)
04-23 10:32:20.739 11549 11549 F DEBUG   :       #14 pc 0000000000084b6c  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: cf739dbc84bcc78f7a1500721bfb3758)
04-23 10:36:44.917 19792 19869 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 19869 (glean.dispatche), pid 19792 (AndroidUI)
04-23 10:36:45.219 19992 19992 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
04-23 10:36:45.220 19992 19992 F DEBUG   : Build fingerprint: 'xiaomi/daisy/daisy_sprout:10/QKQ1.191002.002/V11.0.21.0.QDLMIXM:user/release-keys'
04-23 10:36:45.220 19992 19992 F DEBUG   : Revision: '0'
04-23 10:36:45.220 19992 19992 F DEBUG   : ABI: 'arm64'
04-23 10:36:45.221 19992 19992 F DEBUG   : Timestamp: 2022-04-23 10:36:45+0400
04-23 10:36:45.221 19992 19992 F DEBUG   : pid: 19792, tid: 19869, name: glean.dispatche  >>> org.torproject.torbrowser <<<
04-23 10:36:45.221 19992 19992 F DEBUG   : uid: 10154
04-23 10:36:45.221 19992 19992 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
04-23 10:36:45.221 19992 19992 F DEBUG   : Cause: null pointer dereference
04-23 10:36:45.221 19992 19992 F DEBUG   :     x0  000000742481b50a  x1  000000000000008d  x2  000000742481b302  x3  000000742481b680
04-23 10:36:45.221 19992 19992 F DEBUG   :     x4  000000742481b2c8  x5  000000742481b70a  x6  0000000000000000  x7  0000000000000000
04-23 10:36:45.221 19992 19992 F DEBUG   :     x8  0000000000000000  x9  000000744c6f8990  x10 0000000000000000  x11 0000000000000000
04-23 10:36:45.221 19992 19992 F DEBUG   :     x12 0000000000000000  x13 0000000000000000  x14 0000000000000100  x15 0000000000000001
04-23 10:36:45.221 19992 19992 F DEBUG   :     x16 0000007437babdf8  x17 000000753bcab380  x18 0000007424244000  x19 0000000000000047
04-23 10:36:45.221 19992 19992 F DEBUG   :     x20 000000743669baf2  x21 000000000000008d  x22 000000742481b300  x23 000000742481b508
04-23 10:36:45.221 19992 19992 F DEBUG   :     x24 0000007437cbe8e8  x25 000000743663c14c  x26 0000000000000006  x27 0000000000000001
04-23 10:36:45.221 19992 19992 F DEBUG   :     x28 000000743663c164  x29 000000743663c164
04-23 10:36:45.221 19992 19992 F DEBUG   :     sp  000000742481b2f0  lr  0000007434744c44  pc  00000074352021ac
04-23 10:36:45.249 19992 19992 F DEBUG   :
04-23 10:36:45.249 19992 19992 F DEBUG   : backtrace:
04-23 10:36:45.249 19992 19992 F DEBUG   :       #00 pc 00000000048e11ac  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:36:45.250 19992 19992 F DEBUG   :       #01 pc 0000000003e23c40  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
04-23 10:36:45.250 19992 19992 F DEBUG   :       #02 pc 0000000003e23b7c  /data/app/org.torproject.torbrowser-t9AJvJqm2j1R3-pFfJqr4w==/lib/arm64/libxul.so (BuildId: 5497fe61b545c4ca353634356289f0f41d912dc1)
[... cut ...]
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

bekatd, 2022-04-23:
I resolved crash by following google ml kit installation guide (above screenshot).

But this also should be mentioned in docs

Skipperlla, 2022-11-08:
Same problem but it's solution not working for me :/

Skipperlla, 2022-11-08:
Before installing it, was the application opened when running in debug mode?

luatvudinh, 2023-05-31:
After days struggle with this issue. I found that you don't need to add mlkit face detection lib (on the picture above). I did these to resolve the problem.

1. In `app/build.gradle` add `implementation project(':visioncamerafacedetector')`
2. In `settings.gradle` add 
```
include ':visioncamerafacedetector'
project(':visioncamerafacedetector').projectDir = new File(rootProject.projectDir, '../node_modules/vision-camera-face-detector/android')
```
That's all! Hope this will be helpful

Bonus. 
I also use `vision-camera-code-scanner` so I need to add this lines into `settings.gradle` (don't need add tail of the path and `implementation ...` in `app/build.gradle`. Quite different from these two libs. Weird!):
```
include ':visioncameracodescanner'
project(':visioncameracodescanner').projectDir = new File(rootProject.projectDir, '../node_modules')
```

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy