margelo/react-native-vision-camera #1015: 🐛 Caused by: groovy.lang.MissingPropertyException: Could not get unknown property 'android' for task ':react-native-vision-camera:jitsiAndroidSourcesJar' of type org.gradle.api.tasks.bundling.Jar.

praveenbalajia, 2022-04-24:
### What were you trying to do?

I wanted to show the camera preview on the Android platform and facing some build issue 
on iOS it working fine 

### Reproduceable Code

```tsx
Install the Library adding manual installation step

include ':react-native-vision-camera'
project(':react-native-vision-camera').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-vision-camera/android')

in settings.gradle file
```

### What happened instead?

The app is throwing build issues on Android studio Arctic fox 

project info
buildToolsVersion = "30.0.3"
compileSdkVersion = 31
minSdkVersion    = 23
targetSdkVersion = 31
supportLibVersion = "28.0.0"
ndkVersion = "21.4.7075529"
 
classpath 'com.android.tools.build:gradle:4.2.2'
 
distributionUrl=https\://services.gradle.org/distributions/gradle-6.9-all.zip

### Relevant log output

```shell
Caused by: groovy.lang.MissingPropertyException: Cannot get property 'react' on extra properties extension as it does not exist

Caused by: groovy.lang.MissingPropertyException: Could not get unknown property 'android' for task ':react-native-vision-camera:jitsiAndroidSourcesJar' of type org.gradle.api.tasks.bundling.Jar.
```

### Device

emulator

### VisionCamera Version

2.13.1

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-04-24:
Please share both of your `build.gradle` files.

praveenbalajia, 2022-04-25:
[Compressed_build_files.zip](https://github.com/mrousavy/react-native-vision-camera/files/8551772/Compressed_build_files.zip)
@mrousavy Please check ZIP i have shared both of them

mrousavy, 2022-04-25:
Hey - I'm not downloading/opening .zip files from an untrusted source. Please share them via code snippets here (```).

Also, make sure to compare your files with the RN Template (<https://github.com/facebook/react-native/blob/main/template/android/app/build.gradle>) to make sure you didnt remove anything critical. From the error message it looks like something like `project.ext.react = [ ...` is missing in yours

praveenbalajia, 2022-04-25:
Project Gradle content

```
import groovy.json.JsonSlurper
import org.gradle.util.VersionNumber

// Top-level build file where you can add configuration options common to all
// sub-projects/modules.

buildscript {
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:4.2.2'
        classpath 'com.google.gms:google-services:4.3.10'
        classpath 'com.google.firebase:firebase-crashlytics-gradle:2.8.1'
    }
}

ext {
    buildToolsVersion = "30.0.3"
    compileSdkVersion = 31
    minSdkVersion    = 23
    targetSdkVersion = 31
    supportLibVersion = "28.0.0"
    ndkVersion = "21.4.7075529"

    // The Maven artifact groupdId of the third-party react-native modules which
    // Jitsi Meet SDK for Android depends on and which are not available in
    // third-party Maven repositories so we have to deploy to a Maven repository
    // of ours.
    moduleGroupId = 'com.facebook.react'

    // Maven repo where artifacts will be published
    mavenRepo = System.env.MVN_REPO ?: ""
    mavenUser = System.env.MVN_USER ?: ""
    mavenPassword = System.env.MVN_PASSWORD ?: ""

    // Libre build
    libreBuild = (System.env.LIBRE_BUILD ?: "false").toBoolean()

    googleServicesEnabled = project.file('app/google-services.json').exists() && !libreBuild
}

allprojects {
    repositories {
        // React Native (JS, Obj-C sources, Android binaries) is installed from npm.
        maven { url "$rootDir/../node_modules/react-native/android" }
        // Android JSC is installed from npm.
        maven { url("$rootDir/../node_modules/jsc-android/dist") }
        mavenCentral {
            // We don't want to fetch react-native from Maven Central as there are
            // older versions over there.
            content {
                excludeGroup "com.facebook.react"
            }
        }
        google()
        maven { url 'https://www.jitpack.io' }
    }

    // Make sure we use the react-native version in node_modules and not the one
    // published in jcenter / elsewhere.
    configurations.all {
        resolutionStrategy {
            eachDependency { DependencyResolveDetails details ->
                if (details.requested.group == 'com.facebook.react'
                        && details.requested.name == 'react-native') {
[... 548 lines cut ...]
                    dependency.appendNode('groupId', groupId)
                    dependency.appendNode('artifactId', artifactId)
                    dependency.appendNode('version', it.moduleVersion)
                }
            }
        }

    }
    repositories {
        maven {
            url rootProject.ext.mavenRepo
            if (!rootProject.ext.mavenRepo.startsWith("file")) {
                credentials {
                    username rootProject.ext.mavenUser
                    password rootProject.ext.mavenPassword
                }
            }
        }
    }
}

```

To Mention this is jitsi project am doing a customization
https://github.com/jitsi/jitsi-meet

closed 2023-08-18 by mrousavy