margelo/react-native-vision-camera #1016: ❓ How can I know the exact timing of video recording has been started.

Sidebook, 2022-04-25:
### Question

Hi, thank you for this awesome library!

I'm developing an app that supports video recording with a timer and trying to migrate from react-native-camera.
My current implementation is like this:
```
const record = (duration) => {
  camera.startRecording();
  setTimeout(() => {
    camera.stopRecording();
  }, 5000);
}
```

In this way, the actual duration of a recorded video would fluctuate from 3 to 5 seconds.

To set a timer accurately, I need to start a timer when the actual recording happens, not when I call `camera.startRecording()`.

I looked at the implementation bit, but couldn't find an easy way to know the timing. I was imagining like
```
startRecording({
  onRecordingStart: () => { /* start a timer here */ };
})
```

I would like to know if I'm doing something wrong or if there is a technical limitation on this.

I also found a similar issue: https://github.com/mrousavy/react-native-vision-camera/issues/742
The code mentioned in the issue gives me an impression that it's not supported yet: https://github.com/mrousavy/react-native-vision-camera/blob/main/example/src/views/CaptureButton.tsx#L119

Thank you!

### What I tried

Use setTimeout after calling startRecording()

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-04-25:
Hi!

Yeah, I am migrating VisionCamera to TurboModules/Fabric and have a few more options here. Right now I'm limited by RN's architecture, but once we're migrated we can await `startRecording` that should work for you then :)

Sidebook, 2022-04-25:
Okay, thanks!
Maybe a dumb question, do you think it's easy to add another callback similar I mentioned in the description, not await?
```
startRecording({
  onRecordingStart: () => { /* start a timer here */ };
})
```
I'm wondering if I can folk repo and add this callback by myself as a workaround until architecture migration. Measuring accurate time is a core part of my app.

mrousavy, 2022-04-25:
> do you think it's easy to add another callback similar I mentioned in the description, not await?

On the native side callbacks and Promises are currently essentially the same thing. It is not supported to have two for one function, so I need to wait for TurboModules.

I think I'm gonna go with await rather than a callback, purely an API design decision.

mrousavy, 2022-04-25:
As a really dirty workaround you can add a event to the `<Camera>` component (`onRecordingStarted`) that fires when a recording is started, that way you can time better :)

Sidebook, 2022-04-25:
> On the native side callbacks and Promises are currently essentially the same thing.
I see, that makes sense...

Sidebook, 2022-04-25:
> On the native side callbacks and Promises are currently essentially the same thing.

I see that makes sense.

> As a really dirty workaround you can add an event to the <Camera> component (`onRecordingStarted`)

Probably I would go this way. Thank you so much! I'll close this issue.

closed 2022-04-25 by Sidebook

Sidebook, 2022-04-26:
For future readers, here is a repository with `onRecordingStart` prop. It's based on vision camera 2.13.0.

https://github.com/Sidebook/react-native-vision-camera/tree/v2.13.0-start-recording-callback

mrousavy, 2023-11-21:
Started working on something here: https://github.com/mrousavy/react-native-vision-camera/pull/2194

as of now this doesn't really work, but we'll see what we can do