margelo/react-native-vision-camera #1017: How can zoom in IOS ❓

KaushalLalani98, 2022-04-25:
### Question

I am able to zoom in android but unable to zoom in the IOS device's is there any solution for IOS device's.

### VisionCamera Version

2.13.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-04-25:
Incomplete issue. No reproduction steps. No iOS device name. No iOS OS version.

closed 2022-04-25 by mrousavy

mrousavy, 2022-04-25:
Hey I know this is your first issue on GitHub, but you gotta think about my perspective. How do you expect me to help you if I don't have any information? For me everything works, on an iPhone 11.

I have even written a guide for zooming, and the example app is able to zoom.