margelo/react-native-vision-camera #1018: 🐛 Popup Camera still show after unmounted Camera screen

iamvucms, 2022-04-25:
### What were you trying to do?

I want to use front camera device to take a picture

### What happened instead?

After leave camera screen, I still see popup camera still here. It couldn't hide unless I close app.

### Device

VSmart Active 3

### VisionCamera Version

2.10.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-04-25:
Not enough reproduction steps for me.

closed 2022-04-25 by mrousavy

iamvucms, 2022-04-25:
Sorry about missing reproduce code cos I think it's just simple code , I created this issue cos I see my popup camera in keep showing after I close screen having your <Camera/> component:

<Camera
              photo
              ref={cameraRef}
              focusable
              style={styles.cameraView}
              isActive={state.cameraActive}
              device={state.device}
            />
 Camera is working fine, issue is just popup camera didn't hide after unmount

mrousavy, 2022-04-25:
What is _popup camera_?

iamvucms, 2022-04-25:
[image]
That's what I'm refering.

mrousavy, 2022-04-25:
Lol, I don't have a device like this to test. Maybe there is an additional step involved in dismissing this again? Does it also happen if you set `active` to false or completely unmount the camera but stay on the same screen?

iamvucms, 2022-04-25:
Yes I tried to set active to false and keep staying on same screen but the popup camera is still not hiden

Khusnijafar, 2023-07-03:
I have same issue

mrousavy, 2023-11-18:
Hey! I just found out that I really forgot to close and dispose the locked Camera resources, so I just fixed that in this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2174 
Now the Camera fully & synchronously closes all resources (CameraSession, CameraDevice, OpenGL context, Video & Photo outputs, Photo Synchronizer) once the view gets removed from the React view hierarchy ("unmounted"), and things like Flash, Torch, NFC, and other Camera components should be working again.

There is still a small issue that causes once Camera component to turn into a blackscreen when navigating back and forth between two Camera components, that's a pretty rare edge case but I will still try to fix that soon when I have some free time.

If you appreciate my time, expertise and dedication to this project, pleas [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to support the development of this project.