margelo/react-native-vision-camera #1019: button design

LuisPerez-Dev, 2022-04-25:
### Question

How do the design of the buttons for photo come out since it only opens the camera

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-04-25:
Hey sorry I don't understand the question.

mrousavy, 2022-04-25:
Could you maybe show screenshots?

closed 2022-04-25 by mrousavy