margelo/react-native-vision-camera #1020: ❓ Camera size

t0x1c123, 2022-04-25:
### Question

Hi I have a question. I need fit camera in smaller rectangle with same ratio as full size. Now when I style camera I can get any size but it just cut preview but image is in full size. 

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-04-25:
Hey, sorry I don't understand your question. Could you maybe show screenshots?

t0x1c123, 2022-04-25:
Yes sure. So on this screenshot if I make a photo that photo would be exactly like on screenshot. But when I use this library and set for example style with width and height 80% it only cuts the full camera preview and if I make a photo the result would be much bigger (full sized). So I need to set width and height but keep the ratio of preview not only cut it visually. I just don't want to have preview of photo full screen size. I want to make the size like you can see on screenshot but keep ratio of preview. 

[image]

sandeepsingh-dev, 2022-12-08:
any updates on this?

404-html, 2023-02-05:
I've investigated that a bit and looks like the camera is getting weirdly scaled if `<Camera>` element's ratio differs from actual frame ratio. Here's a workaround that worked for me:
```js
const [dynamicHeight, setDynamicHeight] = useState<number | string>(width);
const frameProcessor = useFrameProcessor((frame) => {
   'worklet';
   runOnJS(setDynamicHeight)('100%');
});

return <Camera
   frameProcessor={frameProcessor}
   style={{ width, height: dynamicHeight }}
   ...
/>

```

So basically the hack is: start with square camera component (with desired width) and as soon as 1st frame is getting processed adjust the height.

Chugunenok, 2023-03-01:
> I've investigated that a bit and looks like the camera is getting weirdly scaled if `<Camera>` element's ratio differs from actual frame ratio. Here's a workaround that worked for me:
>
> ```js
> const [...]

Thanks for this great solution! I use your hack in this way:
`const [dynamicRatio, setDynamicRatio] = useState(1);
const frameProcessor = useFrameProcessor((frame) => {
  'worklet';
  runOnJS(setDynamicRatio)(frame.width / frame.height);
});
...
<Camera
  style={{
    width: desiredWidth,
    height: dynamicRatio * desiredWidth,
  }}
  frameProcessor={frameProcessor}
/>`

FranciscoPDNeto, 2023-06-14:
Hello everyone, are there any other solution for that besides using the frame processor? I'm facing this https://github.com/mrousavy/react-native-vision-camera/issues/1281 when trying to apply the frame processor, and the only solution for that was to downgrade the firebase dependency, which it's not a viable option for us here :(

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

Also Preview now supports `resizeMode` prop

closed 2023-09-30 by mrousavy