margelo/react-native-vision-camera #1023: 🐛 Video output

powerpivar, 2022-04-27:
### What were you trying to do?

I wanted to record a video on Android with video format 640x480 but the video output did not corespondent to camera view input, with video format 1280x720 works fine. 

For IOS works fine in both resolutions.

### Reproduceable Code

```tsx
const format = useMemo(() => {
    if (device) {
      return device.formats.find((item) => item.videoWidth === 640 && item.videoHeight === 480);
    }
    return null;
  }, [device]);
```

### What happened instead?

Instead of getting the same video output as input, the recorded video is smaller than the camera view input.

### Device

Xiaomi Mi 8 Lite

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

powerpivar, 2022-05-12:
Hi @mrousavy, any update here?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy