margelo/react-native-vision-camera #1024: If I want to use more than 1 frame processor plugin how to include it in babel❓

Invictus-Munish, 2022-04-28:
### Question

`module.exports = {
  presets: ['module:metro-react-native-babel-preset'],
  plugins: [
    ['module:react-native-dotenv', {
      moduleName: '@env',
      path: '.env',
      blacklist: null,
      whitelist: null,
      safe: true,
      allowUndefined: true
    }],
    [
      'react-native-reanimated/plugin',
      {
        globals: ['__labelImage'],
        globals: ['__scanFaces'],
      },
    ],
  ]
};
`

### VisionCamera Version

2.130

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-04-29:
`globals` is an array. 

**don't** do:

```
globals: ['__labelImage'], globals: ['__scanFaces']
```

**do:**

```
globals: ['__labelImage', '__scanFaces']
```

closed 2022-04-29 by mrousavy

Invictus-Munish, 2022-04-29:
@mrousavy  thanks 
I had one more about react-native in general if u could help me about, is tensorflow support latest react-native version 0.66, I tried it in expo its working but its not working in react-native cli?
If u have any idea regarding this ?
Thanks for help

AjayParambath, 2023-05-13:
My babel.config file looks like this : 
```
module.exports = {
  presets: ["module:metro-react-native-babel-preset", "module:react-native-dotenv"],
  env: {
    production: {},
  },
  plugins: [
    [
      "@babel/plugin-proposal-decorators",
      {
        legacy: true,
      },
    ],
    ["@babel/plugin-proposal-optional-catch-binding"],
    [
      'react-native-reanimated/plugin',
      {
        globals: ['__scanFaces'],
      },
    ], // NOTE: this must be last in the plugins
  ],
}

```

But I get the same error
###  **"Frame Processor threw an error: Can't find variable: __scanFaces"**

```
 "react": "17.0.2",
 "react-native": "^0.64.2", // migrated from 0.63.3 to 0.64.2 to install vision camera
 "react-native-reanimated": "2.3.0-beta.2", 
 "react-native-vision-camera": "2.13.0",
 "vision-camera-face-detector": "^0.1.7", //  Tried  "0.1.8" and "0.1.7" 

```