margelo/react-native-vision-camera #1026: 🐛 Application crashes on start recording

umar4911, 2022-04-29:
### What were you trying to do?

I simply want to record a video nothing else but the application crashes whenever I trigger startRecording

### Reproduceable Code

```tsx
/* Camera with Capture button Component */
 <Camera
          ref={CameraRef}
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={isActive}
          preset="medium"
          video={true}
          audio={true}
          orientation="portrait"
          torch={supportsFlash ? flash : "off"}
          enableZoomGesture={true}
          onInitialized={() => setIsCameraInitialized(true)}
        />
<CaptureButton style={styles.captureButton} camera={CameraRef} />

/* Capture Button code */

const CaptureButton = (props) => {
  const { style, camera } = props;

  return (
    <>
      <TouchableOpacity
        style={[
          style,
          {
            width: BTN_SIZE,
            height: BTN_SIZE,
            zIndex: 4,
            alignItems: "center",
            justifyContent: "center",
            backgroundColor: "pink",
          },
        ]}
        onPress={() => {
          camera.current.startRecording({
            onRecordingFinished: (video) => console.log(video),
            onRecordingError: (error) => console.error(error),
          });
          setTimeout(() => {
            camera.current.stopRecording();
          }, 5000);
        }}
      />
    </>
  );
```

### What happened instead?

I made this small code for debugging and testing the video component. Whenever I press that capture button, my application crashes without any error in the console

### Device

Android

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

joshualongbrake, 2022-05-17:
Experiencing this as well from a new build.  No stack trace on the javascript side.

ThinhNguyen2002, 2022-07-19:
I have the same issue. @mrousavy  please help.

nica0012, 2022-07-20:
This started to happen to me too after upgrading to RN 0.66. Photos still work but video crashes (based on example)

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy