margelo/react-native-vision-camera #1028: time limit on Frame Processors

LuisPerez-Dev, 2022-05-04:
### Question

You can put a time limit on Frame Processors?

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-05-05:
What does this mean?

mrousavy, 2022-05-05:
You mean how long a Frame Processor Plugin should take to execute? After e.g. 200ms kill the function?

LuisPerez-Dev, 2022-05-11:
yes

mrousavy, 2022-05-23:
No, that doesn't exist as an API. You can configure this yourself on your Frame Processor Plugins.
Why would you want this anyways?

closed 2022-05-23 by mrousavy