margelo/react-native-vision-camera #1031: ❓ Is there any way to get the current frame and display it as a static image on screen

jochenshi, 2022-05-05:
### Question

When use frame processors, i want to get the specific frame which the processor get useful data from, and display it as an image. So, is there anyway to achieve this?

### What I tried

I tried to use `takeSnapshot`, but it has delay and is not exactly the same frame when the processor get data from.

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-05-05:
Yes, the `frame` object is an Image. It is obviously not a high quality image as if you would take a photo, but you can convert it to an image and save to file in your Frame Processor Plugin.

closed 2022-05-05 by mrousavy

jochenshi, 2022-05-05:
@mrousavy And how can i convert the frame to an image?

mrousavy, 2022-05-05:
https://stackoverflow.com/questions/42997462/convert-cmsamplebuffer-to-uiimage