margelo/react-native-vision-camera #1032: 🐛 Unmounting the Camera component does not turn off the torch

Helicosaurus-Rex, 2022-05-05:
### What were you trying to do?

Hi,

I am currently using vision-camera in a project to read QR Codes (using 'vision-camera-code-scanner' for the FrameProcessor).
The project also uses react-navigation.

The user can open a page dedicated to QR Code scanning, which only has the active Camera and a button to toggle the torch.
If the torch is kept on and the user goes back (popping the page out of the navigation current stack), the torch stays on.

The only way to turn it off is then to reopen the page, or minimize the whole application.

### Reproduceable Code

```tsx
import React from 'react';

import {
  TouchableOpacity,
} from 'react-native';

import {Camera, useCameraDevices} from 'react-native-vision-camera';

export default function QrPageExemple(props) {
  const [cameraPermission, setCameraPermission] = React.useState("");
  const devices = useCameraDevices();
  const device = devices.back;

  React.useEffect(() => {
    async function requestPermission() {
      const status = await Camera.requestCameraPermission();
      setCameraPermission(status);
    }

    requestPermission();

    // Cleaning, doesn't work
    return () => {
      setEnableFlashLight(false);
    };
  }, []);

  return (
    <>
      <Camera device={device} torch={enableFlashlight ? "on" : "off"} />
      <TouchableOpacity
        onPress={() => {
          setEnableFlashLight(!enableFlashlight);
        }}
      ></TouchableOpacity>
    </>
  );
}
```

### What happened instead?

The torch stayed on after the Camera component was unmounted. Using a cleaning useEffect() hook to set the torch prop to false has no effect.

### Device

Samsung Galaxy A51 (Android 12), untested on iOS

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-05-05:
Hey! Does this happen on iOS too?

mrousavy, 2022-05-05:
I believe this is a reanimated issue. Can you try downgrading to a Reanimated version that doesn't have layout animations yet?

Helicosaurus-Rex, 2022-05-05:
Testing on iOS is gonna take some time but I'll get back to you once done.

To downgrade, I'd need to go back to 2.2.0 which fails the build.

Helicosaurus-Rex, 2022-05-05:
On a side note, would it be useful to expose the API imperatively for it, like it already exists with focus()?

frankbolviken, 2022-05-06:
I got the same issue. Really annoying :)

hirbod, 2022-05-06:
@mrousavy For once, this error is not due to reanimation :D
I tested that, it does indeed not turn the torch off when you unmount it. But I consider this as a weird UI, since you should really not move away from the screen or unmount while recording is running.

frankbolviken, 2022-05-06:
I'm not recording tho. I'm scanning barcodes, or taking photos. And whenever the barcode is recognized a new screen is popping up, making the screen with the camera unmount. But the camera is still running / torch is still on :)

hirbod, 2022-05-06:
Is your green-light indicator running as well?

mrousavy, 2022-05-06:
@hirbod 

> @mrousavy For once, this error is not due to reanimation :D

We don't know that yet 😉 

@frankbolviken 

Can you check if this gets called?

https://github.com/mrousavy/react-native-vision-camera/blob/8f327267d3189f357a46b0e9217c3aa7b9a8d800/android/src/main/java/com/mrousavy/camera/CameraView.kt#L302-L305

Or this func, which is making sure to unmount/mount accoridngly: https://github.com/mrousavy/react-native-vision-camera/blob/8f327267d3189f357a46b0e9217c3aa7b9a8d800/android/src/main/java/com/mrousavy/camera/CameraView.kt#L277-L291

* If this func **is called**, it is a CameraX issue. Maybe upgrading the latest CameraX library solves it.
* If this func **is not called**, it might be either a React Native issue, a Reanimated Issue (Layout Animations holding a strong ref to a view), or a VisionCamera issue.

hirbod, 2022-05-06:
Yeah, in my case, I know that its not LayoutAnimation, since I removed it fully. But this issue here seems to be Android - might be different.

Helicosaurus-Rex, 2022-05-09:
Tested the same codebase on an Android 5.0, I could reproduce the issue as well. I did not notice anything with a greenlight indicator; do you please have a link to that specific Reanimated issue, so I can dig it and be fully sure it's not related to the bug I'm experiencing?

hirbod, 2022-05-09:
@Helicosaurus-Rex https://github.com/software-mansion/react-native-reanimated/issues/3124
But my findings have been iOS only, but there are some similar reports (at least when using react-native-svg) for android. I am not aware of REA + Camera issues with Android though.

MSaxena92, 2022-05-24:
Having a similar issue on android, the torch stays on even after providing a clean-up function in useEffect when unmounting takes place

annasychugina, 2022-05-25:
It's because of reanimated. These changes https://github.com/software-mansion/react-native-reanimated/pull/3157 fix this problem for me 

react-native-reanimated >= 2.5.0,
react-native-vision-camera latest

MSaxena92, 2022-05-26:
Changes in #3157 are specific to iOS, can't find something for android.

annasychugina, 2022-05-26:
I had problem only on IOS. On android I didn't check

cjanietz, 2022-06-30:
I see this problem on Android as well and also I have by now tried many variants of getting this fixed.
What did no help yet:
- Upgrading reanimated
- Upgrading androidx camera
- Playing around with setting the torch state at different lifecycle events in the Kotlin code

I will check a little more and add any info I discover

mrousavy, 2022-07-04:
Oh - I see - could you shoot me a PR? @cjanietz

ktarasenkowow, 2022-07-19:
> I have found the issue:
> updateLifecycleState is called from onDetachedFromWindow and it assumes the hostLifecycleState is the indicator to update the Camera. However this is only bound to the actual current running host session. Removing the view from the tree will not cause it to be marked as "DESTROYED". If I however force the lifecycle state to be set to DESTROYED from onDetachFromWindow the Camera will be released appropriately and the torch be turned off. I am not sure how this works for anyone correctly right now, but to me it looks like a general issue.
> @mrousavy I would be happy to open a PR if you agree with me

It was helped me=) Look like (in CameraView.kt):
```
override fun onDetachedFromWindow() {
    super.onDetachedFromWindow()
    updateLifecycleState()
    lifecycleRegistry.currentState = Lifecycle.State.DESTROYED
  }
```

mrousavy, 2022-07-19:
But what happens if the view gets re-attached? If we set it to destroyed here, it cannot recover from that. E.g. when minimizing the app and opening it again. Or navigating away with RN Freeze...

ktarasenkowow, 2022-07-19:
> But what happens if the view gets re-attached? If we set it to destroyed here, it cannot recover from that. E.g. when minimizing the app and opening it again. Or navigating away with RN Freeze...

I checked through navigate-back-navigate screen and it work fine.

xulihang, 2022-08-03:
The FrameProcessor is also running after the component is unmounted. Maybe related as well? My project: https://github.com/xulihang/vision-camera-dynamsoft-label-recognizer/tree/main/example

Productivix, 2022-09-13:
hi have the issue , on RN 0.70, 
I call the camera 
`  <Camera
            style={CommonPrxStyles.scanwindow}
            device={device}
            isActive={!scanned}
            frameProcessor={frameProcessor}
            frameProcessorFps={5}
            torch={!scanned ? 'on' : 'off'}
          />`

scanned is a state set by a button (typical toggle state on press)  
On Android Samsung S9 , the torch sets on at press, as the camera (good) , but when press again the camera goes off (well) but the torch stays on (not expected) , untill I place the app on back . When I resume the app the torch stays off. Hope to be clear.
stack : 
`"dependencies": {
    "react": "18.1.0",
    "react-native": "0.70.0",
    "react-native-a-beep": "^1.2.0",
    "react-native-reanimated": "^2.10.0",
    "react-native-vision-camera": "^2.14.1",
    "vision-camera-code-scanner": "^0.2.0"
  },`
Very good project , I remain at disposal for your test /improvments .

whtjs, 2022-10-28:
🎉🎉Hi Guys, I have workaround that android vision-camera make allow torch control by ref🎉🎉

## Allow controlling torch by ref in android
```js
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+EnableTorch.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+EnableTorch.kt
new file mode 100644
index 0000000..4057c33
--- /dev/null
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+EnableTorch.kt
@@ -0,0 +1,9 @@
+package com.mrousavy.camera
+
+import kotlinx.coroutines.guava.await
+
+suspend fun CameraView.enableTorch(status: Boolean){
+    val cameraControl = camera?.cameraControl ?: throw CameraNotReadyError()
+
+    cameraControl.enableTorch(status)
+}
\ No newline at end of file
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt
index 7672b15..e5e1a89 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt
@@ -400,4 +400,15 @@ class CameraViewModule(reactContext: ReactApplicationContext) : ReactContextBase
       promise.reject("NO_ACTIVITY", "No PermissionAwareActivity was found! Make sure the app has launched before calling this function.")
     }
   }
+  
+  @ReactMethod
+  fun enableTorch(viewTag: Int,status:Boolean, promise: Promise){
+    coroutineScope.launch{
+      withPromise(promise){
+        val view = findCameraView(viewTag)
+        view.enableTorch(status)
+        return@withPromise null
+      }
+    }
+  }
 }
diff --git a/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h b/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h
index a2ccdcb..597691c 100644
--- a/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h	
+++ b/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h	
@@ -53,7 +53,7 @@ objc_name : NSObject<FrameProcessorPluginBase>
 @end                                                                                \
 @implementation objc_name (FrameProcessorPlugin)                                    \
                                                                                     \
-__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()    \
++(void)load  \
 {                                                                                   \
   [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #name callback:^id(Frame* frame, NSArray<id>* args) {    \
     return [objc_name callback:frame withArgs:args];                               \
diff --git a/node_modules/react-native-vision-camera/src/Camera.tsx b/node_modules/react-native-vision-camera/src/Camera.tsx
index 68417ac..fbfeb64 100644
--- a/node_modules/react-native-vision-camera/src/Camera.tsx
+++ b/node_modules/react-native-vision-camera/src/Camera.tsx
@@ -293,6 +293,19 @@ export class Camera extends React.PureComponent<CameraProps> {
       throw tryParseNativeCameraError(e);
     }
   }
+
+  /**
+   * Control the cameras torch by given status.
+   * @param {Boolean} status
+   */
+  public async enableTorch(status: Boolean): Promise<void> {
+    try{
+      return await CameraModule.enableTorch(this.handle,status)
+    } catch (e){
+      throw tryParseNativeCameraError(e)
+    }
+  }
+
   //#endregion
 
   /**
```

## Usage
```js
   useEffect(() => {
        const listener = BackHandler.addEventListener('hardwareBackPress', function () {
            ref.current.enableTorch(false);
            return false;
        });
        return () => {
            listener.remove();
        };
    }, []);
```

### you can get [react-native-vision-camera+2.15.1.patch](https://gist.github.com/whtjs/c1f9a466ba7286d3a12980cc10882311) file at the link

### how to use .patch file [patch-package](https://www.npmjs.com/package/patch-package)

### Working On

- typescript method auto complete
- supporting ios

#### After finish these jobs i will make a PR, until now you can fix your bug with this patch!! See ya 🎉

mrousavy, 2022-10-28:
Nice work! Seems good to me. 

I will work on a fix soon

whtjs, 2022-10-28:
> Nice work! Seems good to me. 
> 
> 
> 
> I will work on a fix soon

Thank you always for your dedication. Hope i can help you!

Shoaib-ali7860, 2022-11-30:
we can also stop recording before unmounting the screen  
React.useEffect(() => {
    permission();

    return navigation.addListener('beforeRemove', async (e: any) => {
        await stopRecording()
       
      });
  }, []);

Productivix, 2022-11-30:
yes indeed ! and in pure js : 
`   return ()=>
    navigation.addListener('beforeRemove', async (e) => {
      if (torch)  setTorch('off')
    });

  },[navigation,torch])`

with only : 
`<Camera
                ref={cameraRef}
                style={CommonPrxStyles.scanSmallWindow}
                device={device}
                isActive={true} // isActive={!scanned}
                torch={torch}
                photo={true}
              />`

enzzoperez, 2023-01-25:
> 🎉🎉Hi Guys, I have workaround that android vision-camera make allow torch control by ref🎉🎉
>
> ## Allow controlling torch by ref in android
> ```js
> diff --git [...]

Thank so much, it worked for me, but first I had to patch the patch-package 😄  because I had a buffer bug https://github.com/ds300/patch-package/issues/166

I used this patch https://gist.github.com/brussee/e382ed12ca007a88170289e54b526063 but with `maxBuffer: 1024 * 1024 * 1000,`

whtjs, 2023-01-26:
it's glad helped you 😄, and thanks for patch-package bug info

markoprodanovic, 2023-02-08:
Same issue with react-native app running on Samsung Galaxy A12

Unfortunately the patch isn't working for me. Even after applying still seeing:
`cameraRef.current.enableTorch(false) // enableTorch is unknown` 

Tried applying the patch several times and even deleting node_modules and reinstalling from scratch.

whtjs, 2023-02-08:
@markoprodanovic can you give some example that can reproduce the issue?

markoprodanovic, 2023-02-08:
@whtjs it was a mistake on my end and a misunderstanding of how to use patch-package. For context, I was overwriting the patch file by running my own `yarn patch-package react-native-vision-camera`. I was also working with `v0.5.2` instead of `v0.5.1`. Once I sorted out those things it worked great! Thank you :)

Would be great to get this in an official release

whtjs, 2023-02-08:
i will notice you by a comment that you can get rid of patch files :)

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy

mrousavy, 2023-11-18:
Hey! I just found out that I really forgot to close and dispose the locked Camera resources, so I just fixed that in this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2174 
Now the Camera fully & synchronously closes all resources (CameraSession, CameraDevice, OpenGL context, Video & Photo outputs, Photo Synchronizer) once the view gets removed from the React view hierarchy ("unmounted"), and things like Flash, Torch, NFC, and other Camera components should be working again.

There is still a small issue that causes once Camera component to turn into a blackscreen when navigating back and forth between two Camera components, that's a pretty rare edge case but I will still try to fix that soon when I have some free time.

If you appreciate my time, expertise and dedication to this project, pleas [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to support the development of this project.