margelo/react-native-vision-camera #1033: Consuming the image after takePhoto on IOS❓

frankbolviken, 2022-05-06:
### Question

Hi,
First of all. Cool library, and the response time on issues/questions seems to be very quick!

Ive ran into a problem where i get a "file dont exist" error on iOS. On Android it's working fine.
The issue is that I have to get a base64 string of the image taken to comply with the backend we are using (cannot cchange that now). So i tried to use rn-fetch-blob to get the files:
```
  const data = await cameraRef.current.takePhoto(options);
  const fs = RNFetchBlob.fs;
  let filePath = `file://${data.path}`;
  if (Platform.OS === 'ios') {
    const arr = filePath.split('/');
    const dirs = fs.dirs;
    filePath = `${dirs.DocumentDir}/${arr[arr.length - 1]}`;
  }
  if (await fs.exists(filePath)) {
    console.log('File exist');
  } else {
    console.log('file dont exist');
  }
  const base64: string = await fs.readFile(filePath, 'base64');
```

I've tried without the specific ios section (using the path when showing a preview seems to work fine tho? :O)
But when I actually try to consume / retrieve the file i can't get it. With our without file:// etc. It just prints out:
```
 file dont exist
 [Error: No such file 'file:///var/mobile/Containers/Data/Application/3DA1AFA6-A449-458D-AA16-17DEF7D36329/Documents/150F3BB1-95C7-436A-85DB-94A92E0FB55E.jpeg']
 ```

Any ideas?:O 

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-05-06:
Huh that's weird, can you confirm that the file actually exists (e.g. by rendering it in a FastImage)? Maybe it's a permissions issue

frankbolviken, 2022-05-06:
oo, that was ultrafast response :O

Actually, right now it just started working if I did this
```
const data = await cameraRef.current.takePhoto(options);
        const fs = RNFetchBlob.fs;
        let filePath = data.path;
        if (Platform.OS === 'android') {
          filePath = `file://${filePath}`;
        }
        // eslint-disable-next-line @typescript-eslint/no-unsafe-assignment
        const base64: string = await fs.readFile(filePath, 'base64');
```

So hold on a bit while I test some more :P

frankbolviken, 2022-05-10:
Seems like it was working ;)

closed 2022-05-10 by frankbolviken