margelo/react-native-vision-camera #1038: Video recording does not match preview ( Not Fullscreen )❓

LefterisL, 2022-05-11:
### Question

Hello,

First of all thank you for the amazing work on this library. I'm trying to record a short video.

I've build the following UI as you can see in the photos I've attached. The camera is initialized like

`
<View style={{ flex: 1, marginTop: 5, justifyContent: "center" }}>
<Camera
            ref={camera}
            style={StyleSheet.absoluteFill}
            device={device}
            isActive={true}
            video={true}
            audio={true}
          />
</View>
`

It's parent is a SafeViewArea and there are 2 more Views with flex:1 below the camera.

The problem is that the recorded video does not match the preview of it. Screenshot 1 is preview, Screenshot 2 is the playback after it's been recorded. 

[image]
[image]

My guess is the camera is recording full height but the preview is not ( ? )

Any advice on what I can do here ? Should I get the parent view height, width and find a format that is as close to it as possible ? Any ideas ?

Thank you

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-05-11:
Hmm, normally preview and video should be in sync. Can you check if the actual recorded file has the same dimensions as you want? Remember that the camera sensor records in a format that works best for it, that does not map to view size.

LefterisL, 2022-05-11:
This is the video on my PC
[image]

Not sure if it's a solution but rendering the video with "resizeMode" to cover seems to be giving the same video as the preview. The video however is way taller.

`<Video
          disableFocus={true}
          repeat={true}
          resizeMode={"cover"}
          style={StyleSheet.absoluteFill}
          source={{ uri: verificationVideo.path }}
        />`

When I switch to resizeMode="contain" it fits in height inside the view so I get black left and right.

I think temporary the "cover" attribute might work.

Is there any way that I can assign specific dimensions for the camera to record ? or at least aspect ration ?

mrousavy, 2022-05-11:
Yea, you can filter formats (`device.formats`) - it's an advanced feature and gives you full control over the resolution setting.

closed 2022-05-11 by mrousavy