margelo/react-native-vision-camera #1039: ❓ Im getting the error: package com.mrousavy.camera does not exist

alphonse92, 2022-05-11:
### Question

Im getting the error: package com.mrousavy.camera does not exist

[image]

### What I tried

Cleaned modules and Following the examples, the guide and other resources

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

zhaleTaheri, 2022-05-15:
I have the same problem

mauro-rogledi, 2022-05-16:
I don't using Expo but I've same problem

alphonse92, 2022-05-16:
I don't know how I solved this, but maybe it can be helpful for yall. After getting this error I opened the project with Android studio. Android Studio started to clean the project and then, I ran `cd android && ./gradlew clean` after that, I was able to run the project.

On the other hand, I removed all manual linkings I made

alphonse92, 2022-05-16:
I'll keep this open because there isn't information about the process of building this library when we run the Project. Or maybe it happen randomly

biskis, 2022-06-21:
And updates here? I have the same error now after cleaning.

zhaleTaheri, 2022-06-21:
My problem was solved
for android
- rm -rf yarn.lock package-lock.json node_modules ios/Pods ios/Podfile.lock android/app/build
- yarn && cd ios && pod update && cd ..
- npx pod-install
- yarn cache clean
- yarn clean:android
- yarn android
for ios
- rm -rf yarn.lock package-lock.json node_modules ios/Pods ios/Podfile.lock android/app/build
- yarn && cd ios && pod update && cd ..
- npx pod-install
- yarn cache clean
- yarn ios

pistonsky, 2022-07-28:
For those who are on Mac.

The issue may appear when you have followed a wrong version of “Getting started” doc during your Mac setup for React Native development. You need to follow the correct one, depending on the version of react native that your project is using.

In my case, I needed to uninstall java, and install it again, but using AdoptOpenJDK, instead of Zulu:
```
brew uninstall --cask zulu11
brew install --cask adoptopenjdk/openjdk/adoptopenjdk11
```
I actually even tried to install java 17, so I had to uninstall both `zulu11` and `zulu17`.

jbrodriguez, 2022-07-29:
i was having this same problem, no amount of cleaning fixed, except applying this patch to the v2.40.0 of the package https://github.com/mrousavy/react-native-vision-camera/pull/1116/files

jasbirrajrana, 2022-08-18:
any fix for this i m getting the exact same issue?

jasbirrajrana, 2022-08-18:
@mrousavy

danielmark0116, 2023-03-19:
> For those who are on Mac.
>
> The issue may appear when you have followed a wrong version of “Getting started” doc during your Mac setup for React Native development. You need to follow the correct [...]

I can confirm that it solved it on my setup as well!

I had zulu's 11 version. I got rid of this one and installed the adoptopen one (used jenv to swithc between versions). Everything works now.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy

DanielWoldehana, 2024-07-23:
@mrousavy @pistonsky I am on react-native 0.73.1 which requires java 17 currently I am using zulu 17 and still getting the same error caused by vision-camera-code-scanner, are there any solutions for this?

IgorDePaula, 2024-12-02:
He created this error on purpose to monetize his component. This component is no longer open source. Consider using another module.