margelo/react-native-vision-camera #1043: 'VisionCamera/VisionCamera.h' file not found🐛

moygospadin, 2022-05-12:
### What were you trying to do?

I trying to build my own frame process plugin. I have the following error could not build module 'VisionCamera',  Bridging-Header make like in example. Trying to set swift version to 5.2. Clean and rebuild all.  Nothing helped
[image]
[image]
 

### What happened instead?

[image]

### Device

XCODE Version 13.3 (13E113)

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

moygospadin, 2022-05-13:
replace content in Bridging-header to this 

#import "FrameProcessorPlugin.h"
#import "Frame.h"

in another files use this imports like upper

closed 2022-05-13 by moygospadin

moygospadin, 2022-05-13:
One more update: main problem was, that i use use_framework, i commented it, and add to necessary pods :modular_headers => true, uncommented use_flipper!. And DONE!

KonstantinZhukovskij, 2022-10-24:
This is a bad decision. For some projects, the use of framework is mandatory. Are there any solutions here?

shakuniyamihir8400, 2022-12-05:
@moygospadin Do you have any other solution instead of commenting **use_framework**,
Please share this here if you have.

rkmackinnon, 2023-04-27:
In case anyone is still looking at this, I have a solution. If you patch the VisionCamera package to create an empty VisionCamera.h file and reference in the podspec as in https://github.com/rodgomesc/vision-camera-code-scanner/blob/4dcfeabd3cd41564257e3ec1b0824f3a39b15014/.yarn/patches/react-native-vision-camera-https-95cda75d0b.patch, that fixes the compile error and the project works with **use_framework**. This patch is based on the newer V3 branch of react-native-vision-camera though so ymmv with the latest V2 release.

huzaifaazim0, 2023-05-12:
Found a solution

The problem is using this use_frameworks! in your PodFile
most people are using use_frameworks! due to firebase v15+ requirement

so if you are using use_frameworks! for any reason
remove it and use :module_headers => true instead for each library due to which you used use_frameworks! earlier

in case of firebase, after remover use_frameworks!
you can add the following line at the end of Podfile

pod 'Firebase', :modular_headers => true
pod 'FirebaseCore', :modular_headers => true
pod 'FirebaseCoreInternal', :modular_headers => true
pod 'GoogleUtilities', :modular_headers => true

This should work fine. now you will be able to use your frame processors.

Hope that helps

bimusiek, 2024-02-09:
For anyone reaching this comment, trying everything written before... remember to install`react-native-worklets-core` plugin and configure it in babel.
https://react-native-vision-camera.com/docs/guides/frame-processors#react-native-worklets-core

This was the culprit in my case.