margelo/react-native-vision-camera #1044: ❓ shutter sound not playing on Android

ParkJungJun, 2022-05-13:
### Question

The camera shutter sound plays in ios, but the shutter sound does not play in Android.

### What I tried

```
<Camera
        ref={camera}
        device={device}
        format={formats[0]}
        isActive={isActive}
       photo={true}
       orientation="portrait"
/>

const result = await camera.current?.takePhoto({
        enableAutoStabilization: true,
        skipMetadata: Platform.OS === 'android',
        qualityPrioritization: 'speed',
        flash: 'off',
});
```

```
buildToolsVersion = "31.0.0"
minSdkVersion = 23
compileSdkVersion = 31
targetSdkVersion = 31
ndkVersion = "21.4.7075529"
```

react-native: 0.68.0

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-05-24:
I don't think there is any default shutter sound effect on Android?

kesteer, 2022-06-05:
@mrousavy 
You are correct in that sense. I think the issue comes from people who are moving over from RN Camera to RNVC, as IIRC RN Camera had a shutter sound implemented for Android. 

A disclaimer in the documentation might be a good idea. Since people testing on both types of devices will probably get thrown off by it. 

Or, I was considering making a PR that would add this feature since there is technically a default sound for this in the Android OS. 
https://developer.android.com/reference/android/media/MediaActionSound#SHUTTER_CLICK

mrousavy, 2022-06-05:
from their docs:

> The sound used by Camera.takePicture to indicate still image capture.

Why doesn't it play then? I think this would be a cool option for VisionCamera. It's configurable both on iOS and Android

kesteer, 2022-06-05:
> Why doesn't it play then? I think this would be a cool option for VisionCamera. It's configurable both on iOS and Android

True, I would assume it'd play anyways. A bit more annecdotal evidence, but in my recent implementation of VisionCamera on Android it also was not playing on my phone when taking a picture. 

Looking back at the old code for RN Camera, it seems they manually called the sound directly instead of using it from `Camera.takePicture`. I wonder if something else is happening here? 

https://github.com/react-native-camera/react-native-camera/pull/1441/files

wilav-dev, 2022-07-26:
any update? i need to reproduce shutter sound

uniquejava, 2022-08-18:
+1, we just switched from RNC to RCVC,   the missing shutter sound makes our client unhappy. 😭

bdobreav, 2022-08-18:
+1, please add shutter sound

mrousavy, 2022-08-18:
Stop spamming +1. If you want me to work on VisionCamera to make _your_ client happy, pay me to do so.

robsonbbs, 2022-08-31:
@mrousavy  Based on the code from RN Camera shared by @kesteer and the fact that android docs also point to `MediaActionSound`, seems like shutter sound should be called manually. I will try to take a look at this next weekend, and open a PR with the fix (possibly based on RN Camera solution).

Keep up the good work, this library is awesome!

2sem, 2023-05-19:
On some devices, flash and shutter is not working.

mrousavy, 2023-08-21:
Hey! I just implemented this in V3! :) https://github.com/mrousavy/react-native-vision-camera/pull/1702

closed 2023-09-01 by mrousavy