margelo/react-native-vision-camera #1045: The Camera doesn't show on iOS simulator, but works on Android emulator❓

squareborg, 2022-05-13:
### Question

Hi, I appear to have an issue with showing / using the camera on ios simulator, ios has permission to use the camera, I have this in the App.tsx
```
 const cameraPermission = await Camera.getCameraPermissionStatus();
      const microphonePermission = await Camera.getMicrophonePermissionStatus();
      switch (cameraPermission) {
        case 'authorized':
          console.log('Camera permission authorized')
          break;
```

and I see `Camera permission authorized` in the console, my camera usage is like so -> 

```
const Camera = ({children, ...rest}) => {
  const devices = useCameraDevices();
  const device = devices.back;
  const camera = useRef<VisionCamera>(null);

  useEffect(() => {
    console.log({device, devices});
  }, [device, devices]);

  if (device == null) {
    return <Text>No camera available</Text>;
  }

  const takePhoto = async () => {
    const photo = await camera.current?.takePhoto();
    console.log({photo});
    if (rest.onPhotoTaken) {
      rest.onPhotoTaken(photo);
    }
  };

  return (
    <View style={{flex: 1}}>
      <VisionCamera
        photo={true}
        ref={camera}
        isActive={true}
        device={device}
        style={{flex: 1}}
        {...rest}
      />
      <Button onPress={() => takePhoto()}>Take Photo</Button>
      <View style={CameraStyles.children}>{children}</View>
    </View>
  );
};
```

and the useEffect comes back 

```{"device": undefined, "devices": {"back": undefined, "external": undefined, "front": undefined, "unspecified": undefined}}```

Any help apprectiated, thanks

[image]

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

moygospadin, 2022-05-13:
ios simulator dont have camera, for testing this functionality use real device instead

gary-cohen, 2022-05-19:
> I asked for the same feature, like react-native-camera had. I solved it by just checking if it was an emulator and instead just using a image.

We implemented this ourselves. We used `react-native-view-shot` and screenshot a reference to a view (with a timestamp) if running on an emulator. Of course I would have preferred it builtin.

mrousavy, 2022-05-24:
Yea the native iOS Camera library does not work on a Simulator, and I didn't have any time (or need) to create that behaviour - I just always test my apps on a real device (is also better for performance plus it's more real use case since the JS Runtime is different from simulator to real device)

closed 2022-05-24 by mrousavy

phazei, 2024-10-31:
Not all of us have a real device, that's why we're building in RN...

mrousavy, 2024-10-31:
Well, it's simply not how it works. If you take a look at the native codebase of VisionCamera (or Camera APIs in general) you will understand.

VisionCamera is a powerful library because it exposes all the camera devices your phone has, and each of it's available camera formats (resolution, fps etc).

A simulator on iOS simply does not have any camera devices at all, hence there is nothing to report. VisionCamera just returns an empty list, because that's how it simply works.

There is no real workaround for this, unless I implement a huge "fake" device implementation, where I give you a "fake" device, with multiple different "fake" formats (resolution, fps etc), that can take "fake" photos (pre-generated image), "fake" videos (needs to have a custom length for start/stop recording, so I will probably have to set up some Metal GPU pipeline to generate random video content using graphics APIs), "fake" code-scanner, "fake" frame processor, and even then people will complain that something doesn't work on the Simulator.
Or people will want to use MacBook Camera as input device. Or custom imported video file. But it just doesn't work this way.

If you want to use Cameras on iOS, you need a real iPhone. If you don't have that, you cannot test it.

You can test your app by running it on your macOS as a macOS app, this has access to your MacBook's facetime camera.