margelo/react-native-vision-camera #1047: 🐛 Metro crashing on every build with ENOENT error

TatuUlmanen, 2022-05-13:
### What were you trying to do?

When running `npx react-native run-android`, the build process starts normally and spawns Metro bundler into another window. The build processes normally, until the task :react-native-vision-camera:compileDebugKotlin is reached. The build continues, but at this point Metro crashes with the error message below:

```
C:\Users\myname\myapp\app\node_modules\metro-hermes-compiler\src\emhermesc.js:77
          throw ex;
          ^

Error: ENOENT: no such file or directory, watch 'C:\Users\myname\myapp\app\node_modules\react-native-vision-camera\android\build\kotlin\compileDebugKotlin\caches-jvm'
    at FSWatcher.<computed> (node:internal/fs/watchers:244:19)
    at Object.watch (node:fs:2264:34)
    at NodeWatcher.watchdir (C:\Users\myname\myapp\app\node_modules\metro\node_modules\jest-haste-map\build\watchers\NodeWatcher.js:156:24)
    at C:\Users\myname\myapp\app\node_modules\metro\node_modules\jest-haste-map\build\watchers\NodeWatcher.js:279:16
    at FSReqCallback.oncomplete (node:fs:199:5) {
  errno: -4058,
  syscall: 'watch',
  code: 'ENOENT',
  path: 'C:\\Users\\tatuu\\jemma\\app\\node_modules\\react-native-vision-camera\\android\\build\\kotlin\\compileDebugKotlin\\caches-jvm',
  filename: 'C:\\Users\\tatuu\\jemma\\app\\node_modules\\react-native-vision-camera\\android\\build\\kotlin\\compileDebugKotlin\\caches-jvm'
}

[process exited with code 7 (0x00000007)]
```

From what I understand, Metro is watching some temporary build files in the react-native-vision-camera build folder, and when these are cleaned up, Metro crashes with a "no such file or directory" error.

I didn't find any other issues regarding this problem, even though this is the third machine this happens in.

Is this a react-native-vision-camera problem or a Metro configuration issue? How can I run the build process in a way that doesn't trigger the ENOENT error in Metro?

### What happened instead?

Metro crashes.

### Device

Android emulator

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Zero-Xiong, 2022-07-04:
I hit the same issue, but my problem is not related with the package react-native-vision-camera because I dont use. I dont know what's the root cause, please help!

chai00xxdd, 2022-07-15:
any solution?

Martinocom, 2022-07-19:
The only _solution_ I found for now is to just relaunch Metro after this error happens. Pretty annoying but kinda works.

reznik99, 2022-09-05:
I am having this issue as well but the error mentions `'C:\Users\myname\myapp\app\node_modules\react-native-reanimated\...`

I tried everything and I can't get metro not to crash.

sahibprashant, 2022-09-09:
I am facing this same issue, Not finding any proper solution! 

Anyone found solution for this?

reznik99, 2022-09-09:
Solution that worked for me:

1  - Build first:
`npx react-native run-android`
This will crash with the same error but that's ok.

2 - Run metro separately:
`npx react-native start` or `npm start`

This works for me (live reload etc..)

I think the issue is metro starts watching files while the `npx react-native run-android` is still building/accessing files, and the two interfere with each other.

mhackeras, 2022-09-14:
Best solution for now is to run: 
adb reverse tcp:8081 tcp:8081 tcp:8081

So you can now build the app once and you don't need to run it again to connect to the metro. You can start metro later and app will connect to it.

Meijer-A, 2023-02-02:
I'm experiencing the same issue with vision camera version 2.15.4

amitfleek01, 2023-04-13:
**1 - Solution** 

Main reason of this problem is your node version
You have to set the specific version of node which is used to create that project
This can fixed your issue

**2 - Solution**

Install the latest version of node

krmao, 2023-04-16:
- same error

```
RuntimeError: abort(Error: EPERM: operation not permitted, lstat 'E:\Workspace\xxx\example\android\app\build\intermediates\cxx\Debug\6n1z393a\obj\arm64-v8a\libappmodules.so.tmp05deaa0'). Build with -s ASSERTIONS=1 for more info.
    at process.abort (E:\xxx\rn_addtoapp_bridge\example\node_modules\metro-hermes-compiler\src\emhermesc.js:402:15)
    at process.emit (node:events:520:28)
    at processEmit [as emit] (E:\Workspace\xxx\example\node_modules\signal-exit\index.js:199:34)
    at emit (node:internal/process/promises:133:20)
    at processPromiseRejections (node:internal/process/promises:260:27)
    at processTicksAndRejections (node:internal/process/task_queues:97:32)
error Command failed with exit code 1.
```

krmao, 2023-04-16:
> * same error
>
> ```
> RuntimeError: abort(Error: EPERM: operation not permitted, lstat [...]

### already fix the error
> metro.config.js
```
// noinspection JSUnresolvedFunction,JSCheckFunctionSignatures,JSUnusedGlobalSymbols,NpmUsedModulesInstalled

const path = require('path');
const escape = require('escape-string-regexp');
const exclusionList = require('metro-config/src/defaults/exclusionList');
const pak = require('../package.json');
const root = path.resolve(__dirname, '..');
const modules = Object.keys({ ...pak.peerDependencies });

const originBlackListRE = exclusionList(modules.map((m) => new RegExp(`^${escape(path.join(root, 'node_modules', m))}\\/.*$`)));

//region for widows RuntimeError: abort(Error: EPERM: operation not permitted, lstat 'E:\Workspace\xxx\example\android\app\build\intermediates\cxx\Debug\6n1z393a\obj\arm64-v8a\libappmodules.so.tmp05deaa0'). Build with -s ASSERTIONS=1 for more info.
// https://stackoverflow.com/a/9213411/4348530
// https://github.com/mrousavy/react-native-vision-camera/issues/1047#issuecomment-1510030983
// https://github.com/expo/expo-cli/issues/2021#issuecomment-1239989633
const extraBlackListRE = new RegExp("(.*\\android\\.*|.*__fixtures__\\.*|node_modules\\\\react\\\\dist\\\\.*|website\\node_modules\\.*|heapCapture\\bundle\.js|.*__tests__\\.*)$");
const finalBlackListRE = new RegExp("(" + originBlackListRE.source + ")|(" + extraBlackListRE.source + ")")
console.log("metro.config.js originBlackListRE=", originBlackListRE)
console.log("metro.config.js extraBlackListRE=", extraBlackListRE)
console.log("metro.config.js finalBlackListRE=", finalBlackListRE)
// originBlackListRE /(^E:\\Workspace\\rn_addtoapp_bridge\\node_modules\\react\\.*$|^E:\\Workspace\\rn_addtoapp_bridge\\node_modules\\react-native\\.*$|\\__tests__\\.*)$/
// extraBlackListRE /(.*\android\.*|.*\__fixtures__\.*|node_modules[\\]react[\\]dist[\\].*|website\node_modules\.*|heapCapture\bundle.js|.*\__tests__\.*)$/
// finalBlackListRE /((^E:\\Workspace\\rn_addtoapp_bridge\\node_modules\\react\\.*$|^E:\\Workspace\\rn_addtoapp_bridge\\node_modules\\react-native\\.*$|\\__tests__\\.*)$)|((.*\android\.*|.*\__fixtures__\.*|node_modules[\\]react[\\]dist[\\].*|website\node_modules\.*|heapCapture\bundle.js|.*\__tests__\.*)$)/
// exit();
//endregion

const metroConfig = {
  projectRoot: __dirname,
  watchFolders: [root],

  // We need to make sure that only one version is loaded for peerDependencies
  // So we block them at the root, and alias them to the versions in example's node_modules
  resolver: {
    blacklistRE: finalBlackListRE,
```

Dhiraj-j, 2023-07-12:
Same Error have to start metro again after build

closed 2023-08-18 by mrousavy

FelipeACP, 2023-10-25:
The issue remains on version 3.6.4