margelo/react-native-vision-camera #1049: battery draining🐛

ajmeersha, 2022-05-17:
### What were you trying to do?

i added the active option based on the route focused and app state, but the app still using the camera and it shows the warning message to the user

### Reproduceable Code

```tsx
const isFocused = useIsFocused();
  const isForeground = useIsForeground();
  const isActive = isFocused && isForeground;

 <PinchGestureHandler
              onGestureEvent={onPinchGesture}
              enabled={isActive}>
              <Reanimated.View style={StyleSheet.absoluteFill}>
                <TapGestureHandler onEnded={onDoubleTap} numberOfTaps={2}>
                  <ReanimatedCamera
                    ref={cameraRef}
                    // isActive={isFocused}
                    style={StyleSheet.absoluteFill}
                    device={device}
                    photo={true}
                    isActive={isActive}
                    onInitialized={onInitialized}
                    animatedProps={cameraAnimatedProps}
                    enableZoomGesture={true}
                    onError={onError}
                    orientation="portrait"
                  />
                </TapGestureHandler>
              </Reanimated.View>
            </PinchGestureHandler>
```

### What happened instead?

isActive is working but it draining battery in background

### Device

android 11 (oneplus 8 pro)

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-05-24:
Can you try logging to the console to see if you are actually rendering the camera with `isActive={false}`? Are you using React Freeze

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy