margelo/react-native-vision-camera #1050: 🐛 Ocr and Qr scanner is not working on iOS

luismiguelmena, 2022-05-17:
### What were you trying to do?

Implement QR and Text scanner

### Reproduceable Code

```tsx
const frameProcessorText = useFrameProcessor(frame => {
    'worklet';
    const scannedOcr = scanOCR(frame);
    if (scannedOcr.result.text !== '') {
      runOnJS(setOcr)(scannedOcr.result.text);
    }
  }, []);
const frameProcessorText = useFrameProcessor(frame => {
    'worklet';
    const scannedOcr = scanOCR(frame);
    if (scannedOcr.result.text !== '') {
      runOnJS(setOcr)(scannedOcr.result.text);
    }
  }, []);
```

### What happened instead?

I have implemented the diferents frameProcessor with text ocr and qr scanner. They are working good on android but not in iOS

### Device

Iphone 11

### VisionCamera Version

2.13.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

sumi-svmx, 2022-05-18:
The OCR plugin can be taken from 
https://github.com/aarongrider/vision-camera-ocr/tree/main/example

mrousavy, 2022-05-24:
Hey! This is not an issue with VisionCamera but rather with the Code Scanner plugin you're using. Also you didn't post any relevant log output, how do you expect me to help you?

closed 2022-05-24 by mrousavy

sumi-svmx, 2022-06-10:
@mrousavy had tried this with [vision-camera-ocr](https://github.com/aarongrider/vision-camera-ocr/blob/main/example/src/App.tsx)
using `GoogleMLKit/TextRecognition` `2.2.0`
This is the log that comes with it (iOS version 14.6)

```
VisionCameraOcrExample[1351:9748767] Initialized TensorFlow Lite runtime.
INFO: Initialized TensorFlow Lite runtime.
VisionCameraOcrExample[1351:9748771] [native] VisionCamera.captureOutput(_:didOutput:from:): The Frame Processor took so long to execute that a frame was dropped.
VisionCameraOcrExample[1351:9748765] [native] VisionCamera.invokeOnFrameProcessorPerformanceSuggestionAvailable(currentFps:suggestedFps:): suggestedFps 0.0...
VisionCameraOcrExample[1351:9748765] [native] VisionCamera.invokeOnFrameProcessorPerformanceSuggestionAvailable(currentFps:suggestedFps:): currentFps 5.0...
VisionCameraOcrExample[1351:9748765] [native] VisionCamera.invokeOnFrameProcessorPerformanceSuggestionAvailable(currentFps:suggestedFps:): Frame Processor Performance Suggestion available!
VisionCameraOcrExample[1351:9749054] [native] VisionCamera.captureOutput(_:didOutput:from:): The Frame Processor took so long to execute that a frame was dropped.
VisionCameraOcrExample[1351:9749054] [native] VisionCamera.captureOutput(_:didOutput:from:): The Frame Processor took so long to execute that a frame was dropped.
VisionCameraOcrExample[1351:9749054] [native] VisionCamera.captureOutput(_:didOutput:from:): The Frame Processor took so long to execute that a frame was dropped.
VisionCameraOcrExample[1351:9749054] [native] VisionCamera.captureOutput(_:didOutput:from:): The Frame Processor took so long to execute that a frame was dropped.
VisionCameraOcrExample[1351:9749054]
```

A fix that seems to work is to convert it to cgImage in the ocr plugin (not ideal) before passing it to the TextRecognizer as per [this](https://github.com/mrousavy/react-native-vision-camera/issues/800#issuecomment-1030107638)

```
let imageBuffer = CMSampleBufferGetImageBuffer(frame.buffer)!
let ciimage = CIImage(cvPixelBuffer: imageBuffer)   
let context = CIContext(options: nil)
let cgImage = context.createCGImage(ciimage, from: ciimage.extent)!
let image = UIImage(cgImage: cgImage)
let visionImage = VisionImage(image: image)
```

sarika-sagare, 2023-04-19:
also have same  error : Frame Processor threw an error:Cant find variable:_scanOCR  .....even if including _scanOCR in babel.config.js