margelo/react-native-vision-camera #1052: 🐛 If you recieved the call while recording, microphone gets muted.

usaidather, 2022-05-18:
### What were you trying to do?

While you are recording the video and if you receive a call, in iOS notification pops up and still if you ignore the call and you are recoding the video, the microphone get muted also even if you ignored the call and continue recording then you won't record the audio.

Any possible solution ?

### What happened instead?

the recorded video has no sound after you recieve the call.

### Device

Iphone 12

### VisionCamera Version

2.13.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

usaidather, 2022-05-20:
@mrousavy any comments ?

mrousavy, 2022-05-23:
Hmm, this should actually work.

usaidather, 2022-05-23:
No its not, If you can verify at your end as well.

mrousavy, 2022-05-23:
Yes, I believe you that it's not, I'm saying that I tested this behaviour and it worked. I don't have any free time to test if it works right now.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy