margelo/react-native-vision-camera #1053: 🐛  onInitialized seems to be called too early on Android

gary-cohen, 2022-05-19:
### What were you trying to do?

On some Android devices (Google Pixel 2) the camera returns from `onInitialized` yet doesn't seem ready to capture. If I put a `setTimeout` of 500ms before allowing `takePhoto` to be called it helps on the faster devices but not on the Pixel 2. I can continue to tweak the timeout, but I'm hoping you see what the issue could be.

### What happened instead?

The camera doesn't take a photo, or when running in Waldo, the screen stays black (see below, the upper row is using `react-native-camera` and the second row is `react-native-vision-camera`).
[image]
 

### Device

Pixel 2, Galaxy S10

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-05-24:
Huh that's interesting, I'm not sure if there is a native event that we need to wait for here: https://github.com/mrousavy/react-native-vision-camera/blob/83943da112fdbaf284dc2f925283754223eda9cf/android/src/main/java/com/mrousavy/camera/CameraView.kt#L499-L508

gary-cohen, 2022-05-24:
@mrousavy I'm not sure exactly (your code seems to look ok). I did see this which could be useful to know when the preview is streaming (which might be the event we could key off of). https://stackoverflow.com/questions/69401931/android-how-to-get-a-callback-in-camerax-when-the-camera-device-has-finished-op

gary-cohen, 2022-08-06:
@mrousavy just checking up with this... do you think we could get an event with the preview is streaming? I think that will do what I'm looking for.

mrousavy, 2022-08-07:
I think that's implied with a Frame Processor, I'll have a look if that also works if the FP use case is not attached...

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

gary-cohen, 2023-08-18:
Thanks @mrousavy. I'll try to find some time to integrate the new code in the coming days/weeks. Thanks again for all of your work on this. 🙏

closed 2023-09-01 by mrousavy