margelo/react-native-vision-camera #1056: How to access image date for current frame of camera?❓

tomyates, 2022-05-22:
### Question

In my app I want to show a masked version of the camera view (in this case a hexagon), however, the only way I can use an SVG mask is with an <Image>. I can't mask the Camera View - it just appears over the SVG mask, which is fine.

I am trying to get a constant stream of image data so I can update a static image rapidly, to give the illusion that's is a camera view.

### What I tried

- I can't use  takePhoto() as it sets off the camera shutter sound and isn't quick enough.
- I could use a screen capture plugin to capture the image data from the Camera view, but this seems expensive waste of resources.

I'm trying to use a frame processor, I assumed that the frame object would contain image data, but it doesn't seem like it does? Anyone know how I can access the data from the frame processor?
I thought frame.toString() would get me the byte or base64 data but it doesn't.

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-05-23:
Hi!

> I thought frame.toString() would get me the byte or base64 data but it doesn't.

That would be an expensive waste of resources.

Why don't you just use react-native-masked-view?

tomyates, 2022-05-23:
@mrousavy Thanks Marc! Thats is a very good question - I'll shall give it a go and report back.
Thanks

tomyates, 2022-05-23:
@mrousavy That worked first time - I have NO idea why I didn't try that first - I even went down the road of using Web GL at one point. Thanks again.

closed 2022-05-23 by tomyates

AmrElsersy, 2023-06-20:
How can I access frame data ?? In a different scenario that dosn't use masking