margelo/react-native-vision-camera #1059: ❓ Resizing images

frankbolviken, 2022-05-24:
### Question

Whats the recommended approach of scaling the images down to optimize storage space etc before uploading to a backend service? We are taking pictures to confirm the placement of packages, but we dont need to have 2MB sized pictures for that.

I've tried playing with presets and formats, but I'm not really getting the results I would expect.

### What I tried

<Camera
    preset={"low"} // 'medium', 'photo' etc

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

frankbolviken, 2022-05-24:
For instance when I completely remove preset, it defaults to an image around 2mb in size with 1920x2560 image size. But if i add a format which has photoHeight of 720 and width of 1080, the image size is 2976x2976. So.. I am confused :O

mrousavy, 2022-05-24:
Hey! Resizing Images/Compressing them is not really the scope of a Camera library. If you want to capture lower quality images, choose a `format` that has lower resolution. Maybe the pixel format also comes into play, and `takePhoto` has a `quality` parameter for JPEG compression as well (e.g. `0.6` for high compression, aka low quality)

closed 2022-05-24 by mrousavy

mrousavy, 2022-05-24:
Huh, you're right. Maybe we can add it here: https://github.com/mrousavy/react-native-vision-camera/blob/83943da112fdbaf284dc2f925283754223eda9cf/ios/CameraView%2BTakePhoto.swift#L56-L110

jbrodriguez, 2022-08-18:
i checked this issue on the android side and i was able to set the quality of the jpeg ([based on this](https://developer.android.com/reference/androidx/camera/core/ImageCapture.Builder#setJpegQuality(int)))

```kotlin
      val imageCaptureBuilder = ImageCapture.Builder()
        .setTargetRotation(outputRotation)
        .setCaptureMode(ImageCapture.CAPTURE_MODE_MINIMIZE_LATENCY)
        .setJpegQuality(30) <-----------------
```
i might take a stab at a pr, although i'm not sure how more complex the ios side could be

jbrodriguez, 2022-08-18:
ok, in ios it looks like we could set the quality here 

```
      var format: [String: Any]?
      // photo codec
      if let photoCodecString = options["photoCodec"] as? String {
        guard let photoCodec = AVVideoCodecType(withString: photoCodecString) else {
          promise.reject(error: .parameter(.invalid(unionName: "PhotoCodec", receivedValue: photoCodecString)))
          return
        }
        if photoOutput.availablePhotoCodecTypes.contains(photoCodec) {
          format = [AVVideoCodecKey: photoCodec]
          if photoCodec == AVVideoCodecTypeJPEG {
            format?[AVVideoQualityKey] = 0.3
          }
        } else {
          promise.reject(error: .capture(.invalidPhotoCodec))
          return
        }
      }
```

haven't tested it yet, but does it look good to you @mrousavy  ?

jbrodriguez, 2022-08-18:
one more thing, since `setJpegQuality ` can only be called as part of `ImageCapture.Builder()` it looks like for this to work a prop like `jpegQuality` should be added to `<Camera ...>` as opposed to the `takePhoto` function

JoeDareZone, 2023-10-19:
I ended up passing the photo to Expo-Image-Manipulator, which worked well.

mrousavy, 2023-10-19:
Yup, it's a good idea to use an external lib for full resizing features

robsonbbs, 2023-10-25:
> I ended up passing the photo to Expo-Image-Manipulator, which worked well.

I am using this approach as well, but I noticed that it adds a lot more time to the capture cycle if you compress Images right after being shot. I am thinking about using it before syncing images with my backend, but I feel that it would just mask the real problem (that is expo-image-manipulator being too slow).