margelo/react-native-vision-camera #1062: ✨ Limiting area on which frame processors executes

MSaxena92, 2022-05-26:
### What feature or enhancement are you suggesting?

A prop that defines the area on which the frame processors will do their processing would be nice. Eg: `rectOfInterest={{ x: 0.25, y: 0.25, width: 0.5, height: 0.5 }}`

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

The now deprecated library react-native-camera had `rectOfInterest ` prop (iOS only though) that would do something similar

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2022-05-27:
Hey! That's not really how this works, because this is an added processing cost to the Frame Processor pipeline. I would need to create a new pixel buffer, copy the original, crop the area, then move on with Frame Processing.

This is a great idea for a custom Frame Processor Plugin, e.g. something like `vision-camera-crop-plugin`, then use it like:

```jsx
const frameProcessor = useFrameProcessor((frame) => {
  'worklet'
  const cropped = cropFrame(frame, { x: 0.25, y: 0.25, width: 0.5, height: 0.5 })
  // do other processing
  const detectedFaces = detectFaces(cropped)
}, [])
```

VisionCamera's motivation is to have a lean core to be easily maintainable and extend on the plugin concept, so I think this feature should not be added to the core.

closed 2022-05-27 by mrousavy

JBuijgers, 2022-06-20:
Making changes to an Image(Proxy) or creating a new Image in a normal way doesn't seem possible. Would it be an idea to have the Android callback receive some kind of Frame object that is equal-ish to the iOS side? Mostly due to the iOS side working with a buffer that can be modified/overwritten. This will most probably have (quite) some impact on performance though (for the Android side).

Just to clarify what I'm trying to achieve: I thought it would be awesome to use [Spectrum](https://github.com/facebookincubator/spectrum) as a frameprocessor plugin. The idea would be to return a new Frame with a modified buffer from the callback method in a frameprocessor (this would need some additional JSI conversion). The new Frame object can then be used by other frameprocessor plugins (e.g. for ML(Kit) purposes).

Hope my story makes some sense @mrousavy, wondering what your thoughts on this are. FYI, I'm quite new to JSI/JNI, but I'd love to give this a go, will take some time though.

mrousavy, 2022-06-20:
Yea, I don't know how to create a new ImageProxy instance (or one of it's actual implementations) either, would have to research that. But I'm sure there's a way, maybe I need to have a different return type, like `Image`? Could you create that? E.g. from a `Bitmap` maybe - I'd just have to add the JSI bindings

JBuijgers, 2022-06-20:
I was looking into creating an Image as well, since ImageProxy is just a simple wrapper, but as far as I could see you always need a HardwareBuffer or a GraphicBuffer. After some digging around I mostly ended up in very complicated (for me ;P) [mazes of C++ code](https://android.googlesource.com/platform/frameworks/base/+/master/media/jni/) to read from mostly hardware devices. Finding [this piece of code](https://android.googlesource.com/platform/frameworks/base/+/master/media/jni/android_media_MediaCodec.cpp#2061) is the closest I got to finding something usable there. That also prompted me to have a look at how iOS does this, and yeah, there's a bit of extra freedom there due to Frame.{h,m} having a single buffer that can be changed, hence the idea to also do this for Android.

I'd happily poke around in the Android rabbit hole some more to see what I can do, but the lack of documentation on how to create an Image from a buffer from Android just gave me a feeling that whatever solution is found, it might not be very sustainable.

mrousavy, 2022-06-20:
Yea I know what you mean - the C++/Java interop sucks on Android. With VisionCamera I'm directly exposing what I get from the Camera (CameraX - ImageProxy), so anything on top of that would be an abstraction I'd have to write (that you can write in a FP Plugin as well), which might come with a performance cost

yberstad, 2022-11-23:
Hi! 

This looks interesting, has there any update on this? 

Is this a task  that could be suitable for sponsoring?

mrousavy, 2022-11-24:
Hi! As stated before, this is not something I want to implement, as this is not how Frame Processors should work. Instead, you should pass that rect of interest to your FP Plugin and handle the cropping there.

bryanltobing, 2023-06-16:
do we have this already? `vision-camera-crop-plugin`

patras1, 2023-09-02:
Hi , 
Is anybody is working on this?
This plugin will be implemented in so many projects :)
Specially in the use of tflite models .

Thanks

mrousavy, 2023-09-06:
Hey - there's no such thing as _limiting the area_, it needs to be a full resize. I have a WIP branch of a resize plugin using vImage/GPU-acceleration but it doesn't work on Android yet.

mrousavy, 2024-01-16:
Btw., if you use Frame Processors and need to resize the Frame, I just now released [vision-camera-resize-plugin](https://github.com/mrousavy/vision-camera-resize-plugin) - a plugin to resize Frames to any size and convert them to any RGB layout (RGB, BGR, ARGB, BGRA, ...)! 🥳