margelo/react-native-vision-camera #1064: 🐛 Camera formats are totally wrong and broken

bekatd, 2022-05-27:
### What were you trying to do?

I want to adjust coordinates returned from frame processor plugin to draw points on screen correctly

### What happened instead?

Camera formats and everything related to it are completely wrong and broken.

- width and heights are swapped,
- unable to set desired formats,
- also `photoWidth` and `photoHeight` of `useCameraFormat` hook are totally wrong, since the width and height of frame in frameprocessor plugin is for example 640x640 while current cameraformat is 2000x4000 :| 
- also camera feed strangely zooms on different formats
- also if you log all available formats and look at the format returned by default, you will discover that it can't be found in the available formats :|
- also on every different device everything works differently and wrongly :|

how all of this is even possible :|

### Device

Android devices

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bekatd, 2022-05-31:
You are absolutely right and sorry for my sort-of harsh expression. I would be working on this too, but I doubt my knowledge of native code will be sufficient for anything :(

garmoshka-mo, 2023-04-07:
Photo & video sizes not configurable neither with presets nor with format props. (With format you can actually change size - but it will be random, not even close to width/height of specified format).

I had to hardcode desired resolution with patch: https://github.com/mrousavy/react-native-vision-camera/files/11178551/react-native-vision-camera%2B2.15.4.patch

mrousavy, 2023-04-07:
Hey @garmoshka-mo! That patch on iOS is not safe and might crash on a few devices.

On Android; yes, I left a few comments in the codebase warning that this is not accurate on Android due to CameraX's design. I will rewrite the android part to Camera2 in the [V3 PR](https://github.com/mrousavy/react-native-vision-camera/issues/1376), sponsor me on GitHub to support this effort. I need a lot of free time to do that.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy