margelo/react-native-vision-camera #1065: 🐛

fluxgenic, 2022-05-28:
### What were you trying to do?

I am trying to integrate vision camera in my react native app. The step tried

1. installed "npm i react-native-vision-camera"
2. change the sdk changes to support vision camera
       buildToolsVersion = "30.0.2"
        minSdkVersion = 26
        compileSdkVersion = 31
        targetSdkVersion = 31
        ndkVersion = "21.4.7075529"
3. run the app in command to $  react-native run-android
4. Build succeeded but app lunch and closed automatically. 
may I know anything i missing from the configuration.  

### What happened instead?

The app get crashed and closed automatically but build success.

### Device

android simulator  AOSP on IA Emulator

### VisionCamera Version

"react-native-vision-camera": "^2.13.3"

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-05-31:
Hi!
I cannot help you with your issue. You did not share any logs with me.

closed 2022-05-31 by mrousavy

fluxgenic, 2022-05-31:
which logs you need. from where i need to take it from?

mrousavy, 2022-05-31:
https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting

Under **Android**, see point 6. I need adb logs

onyii5119, 2023-01-28:
I have tried to install react-native-vision-camera in RN 0.7x but ran into build issues. My question is: can react-native-vision-camera be used with RN 0.7x?