margelo/react-native-vision-camera #1069: Front camera returns null metadata🐛

NachoGarciaAitaca, 2022-05-30:
### What were you trying to do?

I was trying to take a picture with the `takePhoto()` method without skipping metadata. With the back camera it returns a normal metadata but with the front facing camera it gives a null metadata.

### Reproduceable Code

```tsx
const takePhoto = async =>{
    const photo = await cameraRef.current.takePhoto({})
  }
```

### What happened instead?

The camera gives this null metadata:

`{"Orientation": 0, "{Exif}": {"ApertureValue": 0, "BrightnessValue": 0, "ColorSpace": 1, "DateTimeDigitized": null, "DateTimeOriginal": null, "ExifVersion": null, "ExposureBiasValue": 0, "ExposureMode": 0, "ExposureProgram": 0, "ExposureTime": 0, "FNumber": 0, "Flash": 1, "FocalLenIn35mmFilm": 0, "FocalLength": 0, "ISOSpeedRatings": [], "LensMake": null, "LensModel": null, "LensSpecification": [], "MeteringMode": 0, "OffsetTime": null, "OffsetTimeDigitized": null, "OffsetTimeOriginal": null, "PixelXDimension": 0, "PixelYDimension": 0, "SceneType": 1, "SensingMethod": 1, "ShutterSpeedValue": 0, "SubjectArea": [], "SubsecTimeDigitized": null, "SubsecTimeOriginal": null, "WhiteBalance": 0}, "{TIFF}": {"DateTime": null, "Make": null, "Model": null, "ResolutionUnit": 0, "Software": null, "XResolution": 0, "YResolution": 0}}`

### Relevant log output

```shell
{"Orientation": 0, "{Exif}": {"ApertureValue": 0, "BrightnessValue": 0, "ColorSpace": 1, "DateTimeDigitized": null, "DateTimeOriginal": null, "ExifVersion": null, "ExposureBiasValue": 0, "ExposureMode": 0, "ExposureProgram": 0, "ExposureTime": 0, "FNumber": 0, "Flash": 1, "FocalLenIn35mmFilm": 0, "FocalLength": 0, "ISOSpeedRatings": [], "LensMake": null, "LensModel": null, "LensSpecification": [], "MeteringMode": 0, "OffsetTime": null, "OffsetTimeDigitized": null, "OffsetTimeOriginal": null, "PixelXDimension": 0, "PixelYDimension": 0, "SceneType": 1, "SensingMethod": 1, "ShutterSpeedValue": 0, "SubjectArea": [], "SubsecTimeDigitized": null, "SubsecTimeOriginal": null, "WhiteBalance": 0}, "{TIFF}": {"DateTime": null, "Make": null, "Model": null, "ResolutionUnit": 0, "Software": null, "XResolution": 0, "YResolution": 0}}
```

### Device

OnePlus 8T KB2003

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-05-31:
Interesting! This might actually be null here: https://github.com/mrousavy/react-native-vision-camera/blob/de41f8be83b2701c18a03ba146989b285856c34e/ios/PhotoCaptureDelegate.swift#L57

I wonder why though?

NachoGarciaAitaca, 2022-05-31:
I suppose in Android is this line, in my project is not null:

https://github.com/mrousavy/react-native-vision-camera/blob/de41f8be83b2701c18a03ba146989b285856c34e/android/src/main/java/com/mrousavy/camera/CameraView%2BTakePhoto.kt#L105

I've been trying selecting a specific camera device to see if it solves the issue, but it doesn't. I think the strange behavior is the fact that with the back camera it works but with the fron camera it doesn't.

mrousavy, 2022-05-31:
Ah sorry I thought we were talking about iOS! Hmm, I wonder at which part it goes wrong, could you debug this?

NachoGarciaAitaca, 2022-05-31:
I tj

NachoGarciaAitaca, 2022-05-31:
I think the problem is that: 

`const devices = Camera.useCameraDevices();
 const device = devices.front.devices`

´device´ returns only `["wide-angle-camera"]` I don't know if it's the default camera...

NachoGarciaAitaca, 2022-05-31:
Sorry I closed it by accident :(

closed 2022-06-06 by NachoGarciaAitaca