margelo/react-native-vision-camera #1073: Trying to integrate TFLite as a Frame Processor Plugin❓

iyinoluwamt, 2022-06-01:
### Question

Any tips on how to do this and/or structure this project

### What I tried

I've tried to go off the example project directory but I'm not sure how to implement TFLite in the ExampleProcessorPlugin code

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-06-02:
Hey, sorry I've never used TFLite before - what exactly are you struggling with? What did you try?

felixdittrich92, 2022-06-08:
Hi @iyinoluwamt,

you could convert your model into ONNX format and use:
react-native: https://www.npmjs.com/package/onnxruntime-react-native
or native: https://onnxruntime.ai/docs/get-started/with-java.html   java and objective-c bindings exists

iyinoluwamt, 2022-06-08:
> Hey, sorry I've never used TFLite before - what exactly are you struggling with? What did you try?

These are from my Swift Frame Processor Plugin file. I tried to model it off of "https://github.com/tensorflow/examples/blob/master/lite/examples/object_detection/ios/README.md".

[image]
[image]

[image]
[image]
[image]

My issue is that this returns `[]` on all frames so I'm not sure why no inferences are being made (TFLite has a working Swift example). I can provide more if you need.

iyinoluwamt, 2022-06-08:
> Hi @iyinoluwamt,
> 
> you could convert your model into ONNX format and use: (ensure to pass your dummy input batch size for TF models as None to provide different batch sizes otherwise this will be a fixed value ([tf2onnx](https://github.com/onnx/tensorflow-onnx) does not provide dynamic axes like PyTorch)) react-native: https://www.npmjs.com/package/onnxruntime-react-native or native: https://onnxruntime.ai/docs/get-started/with-java.html java and objective-c bindings exists

Do you have any idea how I could get the frame by frame input for the ONNX model without using native code and just react-native libraries?

mrousavy, 2023-06-22:
See https://github.com/tensorflow/tfjs/issues/7773 :)

mrousavy, 2023-09-30:
Hey - I built a fast C++ / JSI / GPU-accelerated plugin just for this: https://github.com/mrousavy/react-native-fast-tflite 🥳

closed 2023-09-30 by mrousavy