margelo/react-native-vision-camera #1076: ✨ RGB color space for Android

ben186, 2022-06-03:
### What feature or enhancement are you suggesting?

It would be nice to have RGB option for color space since it would be much easier to implement real-time inference (TFLite) with Frame Processor.

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

Currently, I need to convert YUV to RGB manually with RenderScript. However, it has been deprecated since Android 12.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2022-06-13:
> It would be nice to have RGB option for color space

I'm getting that from the native Android camera. If they don't have it, I can't add it 🤷

ben186, 2022-06-13:
Ah, the goods news is that the output format can be set to rgba now since CameraX v1.1.0
https://developer.android.com/reference/androidx/camera/core/ImageAnalysis.Builder#setOutputImageFormat(int)

mrousavy, 2023-09-30:
Hey - unfortunately this worked for some time in V3 (i think 3.0.0) but now doesn't work anymore because YUV is just much more efficient and natively supported.

mrousavy, 2023-09-30:
let's leave this open maybe there's gonna be a future API that solves this

mrousavy, 2024-01-30:
Hey - I recently published [vision-camera-resize-plugin](https://github.com/mrousavy/vision-camera-resize-plugin), which allows you to really efficiently convert Frames to RGB - this is the same method that CameraX also uses, but much more flexible because you can convert to ARGB, RGB, RGBA, BGRA, etc.

closed 2024-01-30 by mrousavy

moghtader, 2024-01-30:
> Hey - I recently published [vision-camera-resize-plugin](https://github.com/mrousavy/vision-camera-resize-plugin), which allows you to really efficiently convert Frames to RGB - this is the same [...]

Hi @mrousavy I'm a big fan! Is there a way to use the resize plugin and then funnel that result into my own FrameProcessorPlugin? I'm trying to get an ONNX version of ML pipelining working and this would be greatly helpful.