margelo/react-native-vision-camera #1077: Is reanimated v3 supported yet❓

jzxchiang1, 2022-06-04:
### Question

I tried using reanimated v3 with react-native-vision-camera v2.11.2.

It generally works but frame processors in particular do not work on Android. (It works on iOS.)

Specifically, I am trying to use vision-camera-code-scanner v0.2.0, which relies on frame processors.

The device logs I'm seeing in Flipper are:

```
Setting new Frame Processor...
Found CameraView!
Adapting Shareable value from function (conversion to worklet)...
Successfully created worklet!
Called Frame Processor callback, but `frameProcessor` is null!
Called Frame Processor callback, but `frameProcessor` is null!
Called Frame Processor callback, but `frameProcessor` is null!
...
Removing Frame Processor...
Frame Processor removed!
```

Nothing crashes, but the `frameProcessor` is null warnings get logged very often (equal to my `frameProcessorFps` value of 5). Did reanimated v3 at HEAD break the frame processor code in this library? Thanks!

### VisionCamera Version

2.11.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-06-13:
Hey! That's interesting, I didn't try REA 3 yet. Let me know if you find anything that might be the cause of this issue!

jzxchiang1, 2022-06-14:
Are there any logs or specifically areas of the codebase I should be looking at? I'm not too familiar with how to debug why `frameProcessor` is null. Thanks!

heroic, 2022-06-20:
Vision camera fails to be build on rn 69. few issues. react-native directory used to have only react-native aar, but now it has hermes aar too. This leads finding `rnAAR` to fail.

If that is fixed the next error comes due to REA no longer containing pre-built aar, this leads extractJNI to fail

mrousavy, 2022-06-20:
Can you share the folder structure on RN <0.68 and RN 0.69 for comparison?

heroic, 2022-06-20:
This is 69
[image]

This is 68
[image]

similarly, REA doesn't have AAR anymore with 3.0.0

heroic, 2022-06-22:
[image]

I was able to get the react native and android AAR paths to work. Couldn't get REA to work. Couldn't find the jni headers you were requiring earlier anymore in REA 3.

mrousavy, 2023-09-30:
Hey - yep REA 3 is supported now! Both in V3 and V2 of VisionCamera

closed 2023-09-30 by mrousavy