margelo/react-native-vision-camera #1081: The front camera returns null metadata🐛

NachoGarciaAitaca, 2022-06-06:
### What were you trying to do?

**ANDROID**

I was trying to take a picture with the method `takePhoto()` and with the front facing camera and it returns this metadata:

```
{
   "Orientation":0,
   "{Exif}":{
      "ApertureValue":0,
      "BrightnessValue":0,
      "ColorSpace":1,
      "DateTimeDigitized":null,
      "DateTimeOriginal":null,
      "ExifVersion":null,
      "ExposureBiasValue":0,
      "ExposureMode":0,
      "ExposureProgram":0,
      "ExposureTime":0,
      "FNumber":0,
      "Flash":1,
      "FocalLenIn35mmFilm":0,
      "FocalLength":0,
      "ISOSpeedRatings":[
         
      ],
      "LensMake":null,
      "LensModel":null,
      "LensSpecification":[
         
      ],
      "MeteringMode":0,
      "OffsetTime":null,
      "OffsetTimeDigitized":null,
      "OffsetTimeOriginal":null,
      "PixelXDimension":0,
      "PixelYDimension":0,
      "SceneType":1,
      "SensingMethod":1,
      "ShutterSpeedValue":0,
      "SubjectArea":[
         
      ],
      "SubsecTimeDigitized":null,
      "SubsecTimeOriginal":null,
      "WhiteBalance":0
   },
   "{TIFF}":{
      "DateTime":null,
      "Make":null,
      "Model":null,
      "ResolutionUnit":0,
      "Software":null,
      "XResolution":0,
      "YResolution":0
   }
}
```

I need the **FocalLength** and the **FocalLenIn35mmFilm** but it is impossible to get it. 
The back camera metadata works perfectly btw.

### Reproduceable Code

```tsx
const photo = await cameraRef.current.takePhoto({});

console.log(photo.metadata);
```

### What happened instead?

It returned this null metadata:

```
{
   "Orientation":0,
   "{Exif}":{
      "ApertureValue":0,
      "BrightnessValue":0,
      "ColorSpace":1,
      "DateTimeDigitized":null,
      "DateTimeOriginal":null,
      "ExifVersion":null,
      "ExposureBiasValue":0,
      "ExposureMode":0,
      "ExposureProgram":0,
      "ExposureTime":0,
      "FNumber":0,
      "Flash":1,
      "FocalLenIn35mmFilm":0,
      "FocalLength":0,
      "ISOSpeedRatings":[
         
      ],
      "LensMake":null,
      "LensModel":null,
      "LensSpecification":[
         
      ],
      "MeteringMode":0,
      "OffsetTime":null,
      "OffsetTimeDigitized":null,
      "OffsetTimeOriginal":null,
      "PixelXDimension":0,
      "PixelYDimension":0,
      "SceneType":1,
      "SensingMethod":1,
      "ShutterSpeedValue":0,
      "SubjectArea":[
         
      ],
      "SubsecTimeDigitized":null,
      "SubsecTimeOriginal":null,
      "WhiteBalance":0
   },
   "{TIFF}":{
      "DateTime":null,
      "Make":null,
      "Model":null,
      "ResolutionUnit":0,
      "Software":null,
      "XResolution":0,
      "YResolution":0
   }
}
```

### Relevant log output

```shell
{
   "Orientation":0,
   "{Exif}":{
      "ApertureValue":0,
      "BrightnessValue":0,
      "ColorSpace":1,
      "DateTimeDigitized":null,
      "DateTimeOriginal":null,
      "ExifVersion":null,
      "ExposureBiasValue":0,
      "ExposureMode":0,
      "ExposureProgram":0,
      "ExposureTime":0,
      "FNumber":0,
      "Flash":1,
      "FocalLenIn35mmFilm":0,
      "FocalLength":0,
      "ISOSpeedRatings":[
         
      ],
      "LensMake":null,
      "LensModel":null,
      "LensSpecification":[
         
      ],
      "MeteringMode":0,
      "OffsetTime":null,
      "OffsetTimeDigitized":null,
      "OffsetTimeOriginal":null,
      "PixelXDimension":0,
      "PixelYDimension":0,
      "SceneType":1,
      "SensingMethod":1,
      "ShutterSpeedValue":0,
      "SubjectArea":[
         
      ],
      "SubsecTimeDigitized":null,
      "SubsecTimeOriginal":null,
      "WhiteBalance":0
   },
   "{TIFF}":{
      "DateTime":null,
      "Make":null,
      "Model":null,
      "ResolutionUnit":0,
      "Software":null,
      "XResolution":0,
      "YResolution":0
   }
}
```

### Device

OnePlus 8T (Android 12)

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-06-13:
Huh this is really weird, apparently the native camera library cannot get that value, hence it's default value of `0`. https://github.com/mrousavy/react-native-vision-camera/blob/fb2156ec397f573fc7beac56de6b8975f5831500/android/src/main/java/com/mrousavy/camera/utils/ExifInterface%2BbuildMetadataMap.kt#L30

I don't think I'm doing anything wrong here, might be an issue with CameraX?

aguedob, 2022-08-03:
Same issue here using a couple of Samsung devices: frontal camera returns null Exif data and Orientation is always 0

ksegla, 2022-09-22:
I have the same issue. Samsung devices too.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy

agrik-monere, 2025-01-03:
I am facing the same issue in v3.6.4... Any updates?