margelo/react-native-vision-camera #1083: 🐛 Execution failed for task ':react-native-vision-camera:configureCMakeDebug'

abdullahIsa, 2022-06-08:
### What were you trying to do?

Hello, I suddenly started getting this error when running the android build.

-  "react": "17.0.2"
-  "react-native": "0.68.1"
-  "react-native-vision-camera": "^2.13.3"

Things I have done

- I deleted my node_modules folder
- I ran npm install
- cd android
- ./gradlew clean
- ./gradlew assembleDebug (Error also happenes here)

### What happened instead?

The build ended up with error.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:configureCMakeDebug'.

### Relevant log output

```shell
C/C++: C:\Users\Abdullah\Desktop\Programming\company\apps\AppName\AppName-0.68.1\node_modules\react-native-vision-camera\android\CMakeLists.txt debug|armeabi-v7a : CMake Warning at C:/Users/Abdullah/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android-legacy.toolchain.cmake:416 (message):
  An old version of CMake is being used that cannot automatically detect
  compiler attributes.  Compiler identification is being bypassed.  Some
  values may be wrong or missing.  Update to CMake 3.19 or newer to use
  CMake's built-in compiler identification.
Call Stack (most recent call first):
  C:/Users/Abdullah/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android.toolchain.cmake:55 (include)
  C:/Users/Abdullah/Desktop/Programming/company/apps/AppName/AppName-0.68.1/node_modules/react-native-vision-camera/android/.cxx/Debug/2h3q1o64/armeabi-v7a/CMakeFiles/3.10.2/CMakeSystem.cmake:6 (include)
  C:/Users/Abdullah/Desktop/Programming/company/apps/AppName/AppName-0.68.1/node_modules/react-native-vision-camera/android/.cxx/Debug/2h3q1o64/armeabi-v7a/CMakeFiles/CMakeTmp/CMakeLists.txt:2 (project)

w: Detected multiple Kotlin daemon sessions at build\kotlin\sessions

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:configureCMakeDebug'.
> C/C++: C:\Users\Abdullah\Desktop\Programming\company\apps\AppName\AppName-0.68.1\node_modules\react-native-vision-camera\android\CMakeLists.txt debug|armeabi-v7a : CMake Error at C:/Users/Abdullah/AppData/Local/Android/Sdk/cmake/3.10.2.4988404/share/cmake-3.10/Modules/CMakeTestCCompiler.cmake:52 (message):
    The C compiler

      "C:/Users/Abdullah/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/bin/clang.exe"

    is not able to compile a simple test program.

    It fails with the following output:

      Change Dir: C:/Users/Abdullah/Desktop/Programming/company/apps/AppName/AppName-0.68.1/node_modules/react-native-vision-camera/android/.cxx/Debug/2h3q1o64/armeabi-v7a/CMakeFiles/CMakeTmp

      Run Build Command:"C:\Users\Abdullah\AppData\Local\Android\sdk\cmake\3.10.2.4988404\bin\ninja.exe" "cmTC_fef06"
      [1/2] Building C object CMakeFiles/cmTC_fef06.dir/testCCompiler.c.o
      [2/2] Linking C executable cmTC_fef06
      FAILED: cmTC_fef06
      cmd.exe /C "cd . && C:\Users\Abdullah\AppData\Local\Android\Sdk\ndk\23.1.7779620\toolchains\llvm\prebuilt\windows-x86_64\bin\clang.exe 
--target=armv7-none-linux-androideabi21 --gcc-toolchain=C:/Users/Abdullah/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/Abdullah/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -march=armv7-a -mthumb -Wformat -Werror=format-security  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--no-undefined -Qunused-arguments 
-Wl,--gc-sections CMakeFiles/cmTC_fef06.dir/testCCompiler.c.o  -o cmTC_fef06  -latomic -lm && cd ."
      clang: error: unable to execute command: program not executable
      clang: error: linker command failed with exit code 1 (use -v to see invocation)
      ninja: build stopped: subcommand failed.

    CMake will not be able to correctly generate this project.
  Call Stack (most recent call first):
    CMakeLists.txt

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.    

See https://docs.gradle.org/7.3.3/userguide/command_line_interface.html#sec:command_line_warnings

Execution optimizations have been disabled for 7 invalid unit(s) of work during this build to ensure correctness.
Please consult deprecation warnings for more details.

BUILD FAILED in 7m 10s
888 actionable tasks: 882 executed, 6 up-to-date
```

### Device

Windows Laptop/Android Phone

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

abdullahIsa, 2022-06-11:
For the time being, I removed react-native-vision-camera but if anyone can help out would be really great as its importance in my project, thanks.

mrousavy, 2022-06-13:
Hmm, maybe clean your gradle cache and try again, maybe also reinstall the CMake plugin/NDK from Android Studio? Looks like a bug in the compiler

abdullahIsa, 2022-06-13:
> Hmm, maybe clean your gradle cache and try again, maybe also reinstall the CMake plugin/NDK from Android Studio? Looks like a bug in the compiler

Hello, thanks for the reply and yes I have done what you stated but still the same error not matter what i do and it only occurs for react-native-vision-camera.

In my Android studio, the highest **CMake is 3.18.1** but in the error logs given it says to use **CMake 3.19 or newer to use**
[image]

### My build.gradle > android\app

````
apply plugin: "com.android.application"

import com.android.build.OutputFile
import org.apache.tools.ant.taskdefs.condition.Os

/**
 * The react.gradle file registers a task for each build variant (e.g. bundleDebugJsAndAssets
 * and bundleReleaseJsAndAssets).
 * These basically call `react-native bundle` with the correct arguments during the Android build
 * cycle. By default, bundleDebugJsAndAssets is skipped, as in debug/dev mode we prefer to load the
 * bundle directly from the development server. Below you can see all the possible configurations
 * and their defaults. If you decide to add a configuration block, make sure to add it before the
 * `apply from: "../../node_modules/react-native/react.gradle"` line.
 *
 * project.ext.react = [
 *   // the name of the generated asset file containing your JS bundle
 *   bundleAssetName: "index.android.bundle",
 *
 *   // the entry file for bundle generation. If none specified and
 *   // "index.android.js" exists, it will be used. Otherwise "index.js" is
 *   // default. Can be overridden with ENTRY_FILE environment variable.
 *   entryFile: "index.android.js",
 *
 *   // https://reactnative.dev/docs/performance#enable-the-ram-format
 *   bundleCommand: "ram-bundle",
 *
 *   // whether to bundle JS and assets in debug mode
 *   bundleInDebug: false,
 *
 *   // whether to bundle JS and assets in release mode
 *   bundleInRelease: true,
 *
 *   // whether to bundle JS and assets in another build variant (if configured).
 *   // See http://tools.android.com/tech-docs/new-build-system/user-guide#TOC-Build-Variants
 *   // The configuration property can be in the following formats
 *   //         'bundleIn${productFlavor}${buildType}'
 *   //         'bundleIn${buildType}'
 *   // bundleInFreeDebug: true,
 *   // bundleInPaidRelease: true,
 *   // bundleInBeta: true,
 *
 *   // whether to disable dev mode in custom build variants (by default only disabled in release)
 *   // for example: to disable dev mode in the staging build type (if configured)
 *   devDisabledInStaging: true,
 *   // The configuration property can be in the following formats
 *   //         'devDisabledIn${productFlavor}${buildType}'
 *   //         'devDisabledIn${buildType}'
 *
 *   // the root of your project, i.e. where "package.json" lives
 *   root: "../../",
 *
 *   // where to put the JS bundle asset in debug mode
 *   jsBundleDirDebug: "$buildDir/intermediates/assets/debug",
 *
 *   // where to put the JS bundle asset in release mode
 *   jsBundleDirRelease: "$buildDir/intermediates/assets/release",
 *
 *   // where to put drawable resources / React Native assets, e.g. the ones you use via
 *   // require('./image.png')), in debug mode
 *   resourcesDirDebug: "$buildDir/intermediates/res/merged/debug",
[... 328 lines cut ...]
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url("$rootDir/../node_modules/react-native/android")
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }
        maven {
            // react-native-background-fetch
            url("${project(':react-native-background-fetch').projectDir}/libs")
        }
        mavenCentral {
            // We don't want to fetch react-native from Maven Central as there are
            // older versions over there.
            content {
                excludeGroup "com.facebook.react"
            }
        }   
        google()
        maven { url 'https://www.jitpack.io' }
    }
}

````

abdullahIsa, 2022-07-17:
Build working now, not fully sure how but i think part of the reason it's working now is due to upgrading react-native from 0.68.1 to 0.68.2

closed 2022-07-17 by abdullahIsa

mrousavy, 2022-07-18:
Hey! I was on vacation, I'm catching up with [Margelo](https://margelo.io) stuff today and if I have some time left I'll check out the issue for you. Maybe I can find a quick fix.

mrinmoygh081, 2022-09-27:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]

I'm facing the same issue. 

"react": "18.0.0",
"react-native": "^0.69.4",
"vision-camera-code-scanner": "^0.2.0"

Please help me to solve the problem.

mrinmoygh081, 2022-09-29:
I solved the issue for now.
Downgrade react-native-vision-camera to 2.14.0",

You will get a new bug.
To solve that, You need to add a patch file. You should follow [#1118](https://github.com/mrousavy/react-native-vision-camera/issues/1118)

scaralfred, 2022-10-16:
Still having the issue.

deepanshu770, 2022-11-10:
having the same issue in mmkv ,build for android

"react": "18.0.0",
"react-native": "0.69.7",
"react-native-mmkv": "^2.5.1"

System:
    OS: macOS 13.0.1
    CPU: (8) arm64 Apple M1
    Memory: 136.14 MB / 16.00 GB
    Shell: 5.8.1 - /bin/zsh
  Binaries:
    Node: 16.16.0 - /usr/local/bin/node
    Yarn: Not Found
    npm: 8.11.0 - /usr/local/bin/npm
    Watchman: 2022.11.07.00 - /opt/homebrew/bin/watchman
  Managers:
    CocoaPods: Not Found
  SDKs:
    iOS SDK:
      Platforms: DriverKit 22.1, iOS 16.1, macOS 13.0, tvOS 16.1, watchOS 9.1
    Android SDK: Not Found
  IDEs:
    Android Studio: Not Found
    Xcode: 14.1/14B47b - /usr/bin/xcodebuild
  Languages:
    Java: 11.0.17 - /usr/bin/javac
  npmPackages:
    @react-native-community/cli: Not Found
    react: 18.0.0 => 18.0.0 
    react-native: 0.69.7 => 0.69.7 
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found

andreacab2, 2023-01-12:
Fixed with solution [here](https://github.com/mrousavy/react-native-vision-camera/issues/1283#issuecomment-1380368472)