margelo/react-native-vision-camera #1084: ❓ Camera View not show in Full Screen

DhrupalGajjar, 2022-06-09:
### Question

How to show camera in full screen in android in Every Device  in some device it's not show in full screen leave space Right or Bottom Side Please Give suggestion for this it's look like this 
[image]
Any solution to resolve it?

### What I tried

I tried "flex:1" in styles as well as "StyleSheet Absolute Fill" and "height width 100%" and also remove parent view also

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-06-13:
Hey, never seen that issue before - but it looks like a layout issue. Does it work if you replace the `<Camera>` with a normal `<View>` and give it a background color?

hebamoustafa, 2022-07-09:
@mrousavy i have the same issue on android 
and yes it work when replace <Camera> with normal <View>

truongngodang, 2022-07-20:
I got this issues https://imgur.com/qUDL4k6

DevZatokio, 2022-08-22:
@truongngodang any solution?

DhrupalGajjar, 2022-08-23:
Yes

DhrupalGajjar, 2022-08-23:
Send me your android manifest file screen shot i will let you know

DevZatokio, 2022-08-23:
<manifest xmlns:android="http://schemas.android.com/apk/res/android" package="xxx">

  <uses-permission android:name="android.permission.INTERNET" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

  <uses-permission android:name="android.permission.CAMERA" />
  
  <uses-permission android:name="android.permission.RECORD_AUDIO" />
  
  <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
  <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

  <application android:requestLegacyExternalStorage="true" android:largeHeap="true" android:hardwareAccelerated="false" android:usesCleartextTraffic="true" android:allowBackup="false" android:name=".MainApplication" android:label="@string/app_name" android:icon="@mipmap/ic_launcher" android:roundIcon="@mipmap/ic_launcher_round" android:theme="@style/AppTheme">
    
    <meta-data android:name="com.facebook.sdk.ApplicationId" android:value="@string/facebook_app_id" />
    <meta-data android:name="com.facebook.sdk.ClientToken" android:value="@string/facebook_client_token" />
    <activity android:name="com.facebook.FacebookActivity" android:configChanges="keyboard|keyboardHidden|screenLayout|screenSize|orientation" android:label="@string/app_name" />
    <activity android:name="com.facebook.CustomTabActivity" android:exported="true">
      <intent-filter>
        <action android:name="android.intent.action.VIEW" />
        <category android:name="android.intent.category.DEFAULT" />
        <category android:name="android.intent.category.BROWSABLE" />
        <data android:scheme="@string/fb_login_protocol_scheme" />
      </intent-filter>
    </activity>
    
    <activity android:name=".MainActivity" android:label="@string/app_name" android:configChanges="keyboard|keyboardHidden|orientation|screenLayout|screenSize|smallestScreenSize|uiMode" android:launchMode="singleTask" android:windowSoftInputMode="adjustResize" android:exported="true">
      <intent-filter>
        <action android:name="android.intent.action.MAIN" />
        <category android:name="android.intent.category.LAUNCHER" />
      </intent-filter>
    </activity>
    
    <meta-data android:name="com.google.android.geo.API_KEY" android:value="@string/maps_sdk" />
  </application>
</manifest>

DevZatokio, 2022-08-23:
Manifest
```
<manifest xmlns:android="http://schemas.android.com/apk/res/android" package="xxx">

  <uses-permission android:name="android.permission.INTERNET" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

  <uses-permission android:name="android.permission.CAMERA" />
  <!-- optionally, if you want to record audio: -->
  <uses-permission android:name="android.permission.RECORD_AUDIO" />
  <!-- storage -->
  <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
  <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

  <application android:requestLegacyExternalStorage="true" android:largeHeap="true" android:hardwareAccelerated="false" android:usesCleartextTraffic="true" android:allowBackup="false" android:name=".MainApplication" android:label="@string/app_name" android:icon="@mipmap/ic_launcher" android:roundIcon="@mipmap/ic_launcher_round" android:theme="@style/AppTheme">
    <!-- facebook -->
    <meta-data android:name="com.facebook.sdk.ApplicationId" android:value="@string/facebook_app_id" />
    <meta-data android:name="com.facebook.sdk.ClientToken" android:value="@string/facebook_client_token" />
    <activity android:name="com.facebook.FacebookActivity" android:configChanges="keyboard|keyboardHidden|screenLayout|screenSize|orientation" android:label="@string/app_name" />
    <activity android:name="com.facebook.CustomTabActivity" android:exported="true">
      <intent-filter>
        <action android:name="android.intent.action.VIEW" />
        <category android:name="android.intent.category.DEFAULT" />
        <category android:name="android.intent.category.BROWSABLE" />
        <data android:scheme="@string/fb_login_protocol_scheme" />
      </intent-filter>
    </activity>
    <!-- end facebook -->
    <activity android:name=".MainActivity" android:label="@string/app_name" android:configChanges="keyboard|keyboardHidden|orientation|screenLayout|screenSize|smallestScreenSize|uiMode" android:launchMode="singleTask" android:windowSoftInputMode="adjustResize" android:exported="true">
      <intent-filter>
        <action android:name="android.intent.action.MAIN" />
        <category android:name="android.intent.category.LAUNCHER" />
      </intent-filter>
    </activity>
    <!-- maps sdk -->
    <meta-data android:name="com.google.android.geo.API_KEY" android:value="@string/maps_sdk" />
  </application>
</manifest>
```

DhrupalGajjar, 2022-08-23:
> Manifest
> 
> ```
> <manifest xmlns:android="http://schemas.android.com/apk/res/android" package="xxx">
> 
>   <uses-permission android:name="android.permission.INTERNET" />
>   <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
> 
>   <uses-permission android:name="android.permission.CAMERA" />
>   
>   <uses-permission android:name="android.permission.RECORD_AUDIO" />
>   
>   <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
>   <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
> 
> 
>   <application android:requestLegacyExternalStorage="true" android:largeHeap="true" android:hardwareAccelerated="false" android:usesCleartextTraffic="true" android:allowBackup="false" android:name=".MainApplication" android:label="@string/app_name" android:icon="@mipmap/ic_launcher" android:roundIcon="@mipmap/ic_launcher_round" android:theme="@style/AppTheme">
>     
>     <meta-data android:name="com.facebook.sdk.ApplicationId" android:value="@string/facebook_app_id" />
>     <meta-data android:name="com.facebook.sdk.ClientToken" android:value="@string/facebook_client_token" />
>     <activity android:name="com.facebook.FacebookActivity" android:configChanges="keyboard|keyboardHidden|screenLayout|screenSize|orientation" android:label="@string/app_name" />
>     <activity android:name="com.facebook.CustomTabActivity" android:exported="true">
>       <intent-filter>
>         <action android:name="android.intent.action.VIEW" />
>         <category android:name="android.intent.category.DEFAULT" />
>         <category android:name="android.intent.category.BROWSABLE" />
>         <data android:scheme="@string/fb_login_protocol_scheme" />
>       </intent-filter>
>     </activity>
>     
>     <activity android:name=".MainActivity" android:label="@string/app_name" android:configChanges="keyboard|keyboardHidden|orientation|screenLayout|screenSize|smallestScreenSize|uiMode" android:launchMode="singleTask" android:windowSoftInputMode="adjustResize" android:exported="true">
>       <intent-filter>
>         <action android:name="android.intent.action.MAIN" />
>         <category android:name="android.intent.category.LAUNCHER" />
>       </intent-filter>
>     </activity>
>     
>     <meta-data android:name="com.google.android.geo.API_KEY" android:value="@string/maps_sdk" />
>   </application>
> </manifest>
> ```

you need to remove android:hardwareAccelerated="false" from this file then after it's working perfectly

truongngodang, 2022-08-26:
> 

I tried this way, its worked. thank you.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy

sinarhimalaya, 2024-08-15:
> > Manifest
> > ```
> > <manifest xmlns:android="http://schemas.android.com/apk/res/android" package="xxx">
> > 
> >   <uses-permission android:name="android.permission.INTERNET" />
> >   <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
> > 
> >   <uses-permission android:name="android.permission.CAMERA" />
> >   
> >   <uses-permission android:name="android.permission.RECORD_AUDIO" />
> >   
> >   <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
> >   <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
> > 
> > 
> >   <application android:requestLegacyExternalStorage="true" android:largeHeap="true" android:hardwareAccelerated="false" android:usesCleartextTraffic="true" android:allowBackup="false" android:name=".MainApplication" android:label="@string/app_name" android:icon="@mipmap/ic_launcher" android:roundIcon="@mipmap/ic_launcher_round" android:theme="@style/AppTheme">
> >     
> >     <meta-data android:name="com.facebook.sdk.ApplicationId" android:value="@string/facebook_app_id" />
> >     <meta-data android:name="com.facebook.sdk.ClientToken" android:value="@string/facebook_client_token" />
> >     <activity android:name="com.facebook.FacebookActivity" android:configChanges="keyboard|keyboardHidden|screenLayout|screenSize|orientation" android:label="@string/app_name" />
> >     <activity android:name="com.facebook.CustomTabActivity" android:exported="true">
> >       <intent-filter>
> >         <action android:name="android.intent.action.VIEW" />
> >         <category android:name="android.intent.category.DEFAULT" />
> >         <category android:name="android.intent.category.BROWSABLE" />
> >         <data android:scheme="@string/fb_login_protocol_scheme" />
> >       </intent-filter>
> >     </activity>
> >     
> >     <activity android:name=".MainActivity" android:label="@string/app_name" android:configChanges="keyboard|keyboardHidden|orientation|screenLayout|screenSize|smallestScreenSize|uiMode" android:launchMode="singleTask" android:windowSoftInputMode="adjustResize" android:exported="true">
> >       <intent-filter>
> >         <action android:name="android.intent.action.MAIN" />
> >         <category android:name="android.intent.category.LAUNCHER" />
> >       </intent-filter>
> >     </activity>
> >     
> >     <meta-data android:name="com.google.android.geo.API_KEY" android:value="@string/maps_sdk" />
> >   </application>
> > </manifest>
> > ```
> 
> you need to remove android:hardwareAccelerated="false" from this file then after it's working perfectly in my app I hope this solution also work for you.

its works thank you