margelo/react-native-vision-camera #1087: Audio Mute or Unmute not proper in video Recording❓

harmit001, 2022-06-09:
### Question

When muted and then later unmuted in the middle section of the playing video, the audio of the entire video plays at the beginning itself, even if it is muted and the audio of the file is over before the video finishes.

For example 

First audio and video both are enabled and after few time audio is disabled and after few seconds audio is enable and stop recording. We can face issue video play properly but All audio play first then mute 

How I can resolve this? 

### What I tried

audio enable and disable

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-06-13:
Sounds like the Master Clock of Audio and Video is somehow not correctly couting pauses. I've seen this issue before, but didn't find a solution back then.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

closed 2023-09-30 by mrousavy