margelo/react-native-vision-camera #1089: 🐛 Illegal callback invocation from native module. This callback type only permits a single invocation from native code

bekatd, 2022-06-11:
### What were you trying to do?

Trying to capture photo, once frame processor returns valid response

### Reproduceable Code

```tsx
const validateFace: TValidateFace = validity => {
  if (validity === 1) {
    faceValidityRate.value += 1;
  } else {
    faceValidityRate.value = 0;
  }
  isGreen.value = withTiming(validity);
  faceWasDetected.value = withTiming(validity === 1 ? 1 : 0);
  onlyOneFaceAllowed.value = withTiming(validity === -1 ? 1 : 0);
  faceValidityRate.value > VALID_FRAME_COUNT && runOnJS(takeSnapshot)();
};

const faceFrameProcessor = useFrameProcessor(frame => {
  'worklet';
  const faces = scanFaces(frame);
  runOnJS(processFrame)({ faces, validateFace });
}, []);

const takeSnapshot = async () => {
  if (camera.current && !isTakingSnapshot.value) {
    isTakingSnapshot.value = true;
    camera.current
      ?.takePhoto({ flash: torchMode })
      .then(data => {
        setTimeout(() => {
          isTakingSnapshot.value = false;
          if (faceValidityRate.value > VALID_FRAME_COUNT) {
            navigation.navigate('CheckFace', { image: data });
          }
        }, 1000);
      })
      .catch((err: any) => {
        isTakingSnapshot.value = false;
        logError(err);
        return err.message || err;
      });
  }
};
```

### What happened instead?

calling `takePhoto()` multiple times (since many frames were valid) caused crash on iOS (Maybe on Android too)

### Device

iPhone 11

### VisionCamera Version

2.13.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bekatd, 2022-06-11:
I solved it by preventing `takePhoto()` to be called multiple times with this:

```
setTimeout(() => {
  if (faceValidityRate.value > VALID_FRAME_COUNT) {
    navigation.navigate('CheckFace', { image: data });
  } else {
    isTakingSnapshot.value = false;
  }
}, 1000);
```

mrousavy, 2022-06-13:
Hi.

> caused crash on iOS

You did not put any logs here. How should I know what happened?

> Maybe on Android too

😅

closed 2022-06-13 by mrousavy