margelo/react-native-vision-camera #1096: ❓ Is there a way to record video with overlay?

da1z, 2022-06-15:
### Question

I want to record video with simple overlay, i want to put speedometer on top of recorded video. is it possible? if it is not supported out of the box maybe you can give me some ideas how it can be implemented? thanks!

### VisionCamera Version

latest

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-06-20:
Hey!

1. The overlay is not part of a Camera library. You can render this overlay yourself, it's just a simple view. If you want to embed it into the actual video output, that's not possible at the moment, you have to create a native module that writes that data to the video after the video has been recorded.
2. At the moment, it is not possible to record videos from two cameras at the same time. This is due to restrictions from the native Android CameraX library.

closed 2022-06-20 by mrousavy

Hamzzay, 2022-12-28:
I'm trying to to add an overlay (image) on the camera and then trying to take the screen of the entire view, it somehow gives a black screen to me.
![Screenshot_20221228-205920_pastiche_app](https://user-images.githubusercontent.com/33403624/209839096-39f46d76-4e41-4524-b798-adb5eefdee76
[image]
.jpg)

Code:

`<ViewShot
            ref={viewShotRef}
            options={{format: 'jpg', quality: 0.9}}
            style={{
              // overflow: 'hidden',
              width: '100%',
              height: '65%',
              flex: 1,
            }}>
            <View style={{flex: 1}}>
              <ReanimatedCamera
                style={{
                  flex: 1,
                  alignItems: 'center',
                  justifyContent: 'center',
                }}
                device={device}
                isActive={true}
                animatedProps={animatedProps}
                enableZoomGesture
                zoom={0}
                torch={flashlight ? 'off' : 'on'}
                photo={true}
              />

              <View style={styles.CarouselView}>
                <Carousel
                  loop
                  width={width}
                  height={300}
                  // autoPlay={true}
                  data={overlays}
                  scrollAnimationDuration={1000}
                  onSnapToItem={index => console.log('current index:', index)}
                  renderItem={({item, index}) => (
                    <View style={styles.scannerView}>
                      <Image
                        source={{uri: decodeURIComponent(item.overlayPath)}}
                        style={styles.scannerImageView}
                      />
                    </View>
                  )}
                />
              </View>
            </View>
          </ViewShot>`

Mourad-Tlili, 2023-08-08:
@Hamzzay Anything new about this topic ?

mrousavy, 2023-08-08:
Yea, see VisionCamera V3