margelo/react-native-vision-camera #1097: 🐛 App crashes on Android: UnsatisfiedLinkError (vision-camera-face-detector)

fleuverouge, 2022-06-16:
### What were you trying to do?

I wanted to detect faces with vision-camera-face-detector.

I tried  `cd android/ && ./gradlew clean && cd .. && watchman watch-del-all && rm -rf $TMPDIR/metro-cache && yarn cache clean && yarn start --reset-cache`

### Reproduceable Code

```tsx
const faceFrameProcessor = useFrameProcessor(
        frame => {
            'worklet';
            const scannedFaces = scanFaces(frame);
            runOnJS(setFaces)(scannedFaces);
        },
        [],
    );
```

### What happened instead?

The app crashes when the screen which contains the Camera view is reached.

### Relevant log output

```shell
FATAL EXCEPTION: pool-33-thread-1

java.lang.UnsatisfiedLinkError: No implementation found for void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy) (tried Java_com_mrousavy_camera_CameraView_frameProcessorCallback and Java_com_mrousavy_camera_CameraView_frameProcessorCallback__Landroidx_camera_core_ImageProxy_2)

FATAL EXCEPTION: pool-33-thread-1
Process: ..., PID: 8529
java.lang.UnsatisfiedLinkError: No implementation found for void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy) (tried Java_com_mrousavy_camera_CameraView_frameProcessorCallback and Java_com_mrousavy_camera_CameraView_frameProcessorCallback__Landroidx_camera_core_ImageProxy_2)
	at com.mrousavy.camera.CameraView.frameProcessorCallback(Native Method)
	at com.mrousavy.camera.CameraView.configureSession$lambda-7$lambda-6(CameraView.kt:491)
	at com.mrousavy.camera.CameraView.$r8$lambda$cqtIchEZdTZaV3R0UUrDpVbB1Es(Unknown Source:0)
	at com.mrousavy.camera.CameraView$$ExternalSyntheticLambda1.analyze(Unknown Source:2)
	at androidx.camera.core.ImageAnalysis.lambda$setAnalyzer$2(ImageAnalysis.java:463)
	at androidx.camera.core.ImageAnalysis$$ExternalSyntheticLambda0.analyze(Unknown Source:2)
	at androidx.camera.core.ImageAnalysisAbstractAnalyzer.lambda$analyzeImage$0$ImageAnalysisAbstractAnalyzer(ImageAnalysisAbstractAnalyzer.java:283)
	at androidx.camera.core.ImageAnalysisAbstractAnalyzer$$ExternalSyntheticLambda1.run(Unknown Source:14)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:764)
```

### Device

Samsung Galaxy A8

### VisionCamera Version

2.13.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-06-20:
Huh that's weird, are you sure you rebuilt the native code correctly?

fleuverouge, 2022-06-22:
Hi @mrousavy I've tried to rebuild the code many times (cleaned cache, removed build folders, deleted node_modules...), also copied build.gradle from the sample project... Without frameProcessor, the camera works normally. The frame processors also run on iOS. The build log stated that frame processor is enabled.

ProkhoDim, 2022-07-13:
> Hi @mrousavy I've tried to rebuild the code many times (cleaned cache, removed build folders, deleted node_modules...), also copied build.gradle from the sample project... Without frameProcessor, [...]

@fleuverouge 
I just got the same error.
I solved it by installing react-native-reanimated and applying the patch-package steps from this [comment](https://github.com/mrousavy/react-native-vision-camera/issues/1118#issuecomment-1175730679) (for react-native 0.69)

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy