margelo/react-native-vision-camera #1098: 🐛 Possible frame processor memory leak on iOS

alexstanbury, 2022-06-17:
### What were you trying to do?

Been investigating memory issues on my app, created a stripped down repo to try and find the culprit through trial and error.

I observed running an empty frame processor on iOS slowly increases memory without it being released.

This effect is multiplied if the frame processor does work using a frame processor plugin and/or calls over to react.

### Reproduceable Code

```tsx
I created a repo here.

`yarn && cd ios && pod install && cd ..`
```

### What happened instead?

Memory increased over time without being released.

### Device

iPhone XR

### VisionCamera Version

2.13.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-06-20:
Huh, interesting! I assume this should be reproduceable in the example app here as well?

Do you notice something similar on Android?

bglgwyng, 2023-03-29:
I think I found the minimal reproducing step. [Here](https://github.com/bglgwyng/RNVC3Sandbox/tree/memory-leak) is the repository.

[As you can see in the repo](https://github.com/bglgwyng/RNVC3Sandbox/blob/memory-leak/App.tsx#L26), `useFrameProcessor` recreates the frame processor on every change of `foo` value. 
I suppose that the object returned by `useFrameProcessor` is not being cleaned up well, and it causes the leak.

The example is with [V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376). However, you can reproduce the leak in the same way with V2 also.

scotjam, 2023-04-28:
I've been trying to write my first react native app using react-native-vision-camera, but my app has been crashing on android without leaving an error message at all (just disappears on the device - no sign in console that it has crashed). I eventually discovered that commenting out the frame processor from the Camera properties in the return value stops it from crashing so I suspect it is the same bug that this issue number refers to. 

Unfortunately I'm trying to do face detection (then use react-native-sound to play a sequence of sounds) so I need the frame processor for the face detection, but it almost never makes it to the end of the sound sequence. I suspect that this bug (1098) may be the same bug I'm facing, but I'm facing it on Android.

I tried reanimated 2.17.0 (as well as the reanimated 2.14.4 that I downgraded to initially, to get around this bug: https://github.com/mrousavy/react-native-vision-camera/issues/1514) and the issue still occurs. But as described in 1514, reanimated 3.x isn't supported by react-native-vision-camera (Property '_setGlobalConsole' doesn't exist) so I don't know if 3.x would fix the problem. 

In the meantime I have had to resort to using a different camera library for face recognition (which is playing nicely with react-native-sound fortunately, hence I'm pretty sure my issue is with using react-native-vision-camera and specifically it is the frame processor's fault, rather than react-native-sound). Would report this as a reanimated bug, but given they're on 3.x now it may have been fixed. 

So I'm posting this comment just so that if there are others who follow the same path I did, they don't get stuck thinking they've done something wrong (I spent hours trying to figure out whether I hadn't release()d a Sound object in some edge case). Unfortunately the current version of react-native-vision-camera relies on react-native-reanimated 2.x, and react-native-reanimated 2.x has memory leak bugs that affect android as well as iOS. 

Looking forward to V3 (without reanimated), and thanks for all your efforts!

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please let me know here!!

closed 2023-09-30 by mrousavy

GreenCurveDev, 2024-09-13:
Hi, I can confirm the issue is still present - running latest app versions and an 'empty' Frame Processor or Skia Frame Processor (with or without calls to 'render') results in leakage of ~0.1MB per second (measured using React Native Dev Menu Perf Monitor). Same leakage occurs when using this in a bare workflow created using an EAS server build deployed directly to a device (in this case an iPhone 12). Left to run with no other interaction for a period of 5-10 mins, all memory is consumed (iP12) and app crashes. Disaster (for me) as discovered in late stages of testing where Frame Processors are being used for simple but continuous augmentation of the camera feed (Saturation, Contrast etc) using shaders. If I could somehow access the camera texture then I could use a package such as expo-GL to achieve the same results without having to switch camera package. For example, expo-gl has a function to create a camera texture from the camera reference. Any suggestions on how to do this using for RNVC would be gratefully received - as it stands Frame Processors cannot be used for this (or perhaps any) use case involving prolonged running of a frame processor function, Thanks & regards, Paul

GreenCurveDev, 2024-09-13:
https://github.com/user-attachments/assets/caacb3e8-2c03-4510-a43d-be7328ee3179

Just to be 100% sure - I downloaded RNVC 4.5.3 to create bare minimum repro. Used two frame Processors (see below) and attached video of memory increase using the first (no Skia frame processor shown below). The memory increase is much more rapid for Skia version but hopefully this illustrates the issue.

const NullFrameProcessor = useFrameProcessor((frame) => {
    'worklet';
    //console.log('Running NULL FP');
  }, [])

and for Skia

const NullSkiaFrameProcessor = useSkiaFrameProcessor((frame) => {
    'worklet';
    //console.log('Running NULL SKIA FP');
    frame.render();
  }, [])

"@shopify/react-native-skia": "^1.3.13", 
"react-native-vision-camera": "^4.5.3",
"react-native-worklets-core": "^1.3.3"

alexstanbury, 2024-09-13:
Interesting - Have you tried on both iOS and Android? Have you tried using a release build and monitoring in Xcode and Android Studio?

mrousavy, 2024-09-13:
Hey! This is interesting. I had Frame Processors running for 30mins without any crashes.

Are you using the Reanimated babel plugin, or worklets-core only?

Pretty sure this will be fixed with [Nitro Modules](https://github.com/mrousavy/nitro) once I rewrite VisionCamera to be one.

GreenCurveDev, 2024-09-13:
Hi, thanks to both.  Just run it via Xcode and monitored memory by enabling Malloc Scrabble etc - this shows the memory increasing in accordance with the behaviour observed using Perf Mon. 
I do have the reanimated plugin enabled but do not use it anywhere in the test code.  Will remove and rerun the test to see if that makes any difference - assume the 30m test was without the reanimated plugin. Will post result shortly.

GreenCurveDev, 2024-09-13:
Hi, I have removed the Babel reanimated plugin and unfortunately the problem remains. When you ran for 30mins was the memory consumption roughly static? Thanks

GreenCurveDev, 2024-09-13:
I think it would take 30+ mins to use all the memory as this is a bare bones app to test the issue - eventually however, I would expect all memory to be consumed and the app to fail.

GreenCurveDev, 2024-09-13:
maybe device settings/formats?

dev-john-nguyen, 2024-10-03:
useFrameProcessor seems to be stable for me, but experiencing frequent crashing with useSkiaFrameProcessor. I posted a discussion about it in react-native-skia https://github.com/Shopify/react-native-skia/discussions/2675. I'm not able to yet pinpoint exactly where the issue is, but seems like gc is not properly releasing memory?

JeremyColfer27, 2025-10-18:
> Actually, with a Skia frame processor and a single call to frame render - memory utilisation reaches 1GB is around 3-4 mins. So there must be something different between your 30min+ test and my own.

Did you ever find answers??